StephenBlackWasAlreadyTaken/xDrip #53: DST change can impact Sensor Age in calibration calculations

jstevensog, 2015-01-29:
When DST changes occur the sensor age alters significantly (either older by 1 hour or younger by one hour).  Likely due to the Sensor Start time being recorded in localtime rather than GMT/UTC.  This should be rectified.  Discovered it when I found my android phone was on the wrong time zone after starting a sensor.  When I set it up correctly and changed the time to match, the sensor age was now 1 hour younger than it was previously.  DST changes will show this up.
Cheers

StephenBlackWasAlreadyTaken, 2015-01-29:
Are you sure? All data datetime data is stored in unix Epoch timestamps so as long as the time is correct (correct for the timezone it is set to) it should have no impact at all.
Changing time zones should also have no impact
The only thing that should have an impact, is changing the actual time

StephenBlackWasAlreadyTaken, 2015-02-25:
closing, I think we discussed this and found the issue, let me know if this is still a thing that needs looking into

closed 2015-02-25 by StephenBlackWasAlreadyTaken