StephenBlackWasAlreadyTaken/xDrip #54: Correct Graph but BG Number and tendency is incorrect

dirkfabisch, 2015-01-29:
The graph is in accordance with the values shown by the meter.
But the number reported above the graph as well as the tendency is not correct

For example:
Graph shows a value round about 100 and is more or less unchanged since an hour.
Nevertheless the number shown above shows HIGH and a double arrow down or up.
(High alarm is set to 270)

StephenBlackWasAlreadyTaken, 2015-01-29:
Hey, these are the predictive values, unfortunately when there is a lot of sensor noise they don't do so well. I'm looking into better ways to determine which points should hold how much weight in determining a good prediction but you are correct that these values won't sound alarms or anything like that, only actual readings will. 

If you pull the latest from master there is an option in settings to turn the predictions off and only display actual readings

StephenBlackWasAlreadyTaken, 2015-01-29:
Also, when you see that check your bg data table to see if there are duplicate readings. Those cause the predictions a lot of trouble

dirkfabisch, 2015-01-29:
I installed already the latest version an disabled the predictions. So the values and tendency must come from the raw values. I will check for duplicate readings, when I get access to the mobile from my wife.

StephenBlackWasAlreadyTaken, 2015-01-29:
oh, with predictions off that would be quite strange, do you use mmol?

dirkfabisch, 2015-01-29:
We use mg

dirkfabisch, 2015-01-29:
Somehow the tick at "Display Predictive Values" was again set. So we take a close look again on the numbers and tendency and I will create a new issue if needed.

closed 2015-01-29 by dirkfabisch