StephenBlackWasAlreadyTaken/xDrip #58: strange looking curve in app..

mayaalinder, 2015-02-02:
I manage to build and install everything.. but when I look at pictures from the comparison of drip vs 505, it looks like the curves in the app are quite "smooth". Mine aren't.. I am not using 505 yet since it is not available in sweden, but I have the old one, and the one in the Animas vibe pump to compare with.. This is what I see that is strange to me.. the values are quite scattered, and look at for example 21.00 it looks like I have two different values at the same time? can this be due to an installation mistake, or due to that I view the result in mmol? The values are not that scattered in the other two..
[image]

raduiliescu, 2015-02-09:
The issue you shown on 21 is easy to reproduce. I have experienced it.
Keep the dexdrip near you, but take the phone out of Bluetooth range or just stop Bluetooth on phone.
Wixel will transmit via UART readings to Bluetooth chip, but the chip won't be able to send them to the phone and will store them.
Once phone is back on range Bluetooth will push all readings.
My plan is to fix this by adding a sequence number on packets send by wixel to the phone.
BgReading has timestamp so just need to figure out timestamp from sequence id.
Special case - if sequence id = 0 means that wixel restarted.

mayaalinder, 2015-02-12:
Let me know when you have it checked in so I can give it a try..

closed 2015-02-25 by mayaalinder