StephenBlackWasAlreadyTaken/xDrip #62: Status screen

raduiliescu, 2015-02-09:
Would be nice to have a status menu/screen accessible via menu.

Info to be put there:
- sensor start time/date;
- last calibration time;
- Bluetooth status;
- wixel battery status/level;
- dexcom transmitter battery status/level;
- last reading time;
- number of reading missed in last hour, last 24 hours, since sensor start;

syntaxerr66, 2015-02-09:
Seems simple enough.  The Wixel battery status and level should be in very soon.  I've been testing it on a branch I have and it seems to work, I'm just calibrating the readings to find the minimum and maximum.    That said, it does require the voltage divider modification.  If you've done that, I can shoot you a link to the branch to try out.

syntaxerr66, 2015-02-12:
I'm going to try and work on a few of these things since I think they're good ideas and seem quick to knock out.  Just a bit of clarification - are you thinking these should be on a screen separate from the home screen?  For the Wixel battery level, the latest PR has this and it's on the home screen above the BG.  I'm going to put the transmitter battery on the home screen as well since this is important info (there's no level for it - it reads either "OK" or "Low battery").

raduiliescu, 2015-02-17:
I think is worth having another screen for this.
Apart from battery level and maybe bluetooth status none of this have what to do on main screen.

playingbball20, 2015-02-22:
syntaxerr66 I would love the code for the battery.  Once I loose the battery, I have to wipe the app to get the dex to be recognized again

StephenBlackWasAlreadyTaken, 2015-02-25:
@playingbball20 you actually wipe all data? it should be scanning again every 3ish minutes for bt while disconnected, I have never had to wipe data... sometimes if I get impaitent I will forceclose the app then it will find it a little faster (still takes a few seconds for android to set up the gatt server, up to a 3 or 4 minutes)

playingbball20, 2015-02-25:
I realized when the battery died, I had to reprogram the wixel.

StephenBlackWasAlreadyTaken, 2015-02-25:
thats even more strange, certainly shouldnt be the case unless your wixel is bad, maybe was just locked in a sleep loop, if you grab the latest wixel code that should not be an issue

playingbball20, 2015-02-25:
I'll make sure I have the newest one, but I thought I did, thanks.  I hope I don't have a bad wixel, I had a bad bt module.

Cagier, 2015-10-15:
I am using an experimental branch on two uploaders and I don't see this.  I have the voltage divider on one using a standard xDrip (not sure what version as that is at home).

And I have 415d549-experimental-2015.09.20 which shows me the Transmitter Battery status on the status screen (and a warning if it Transmitter Battery is low that appears on the main screen).  This uses a "Wifi Wixel" collection method and although I can see the wixel battery data on the database, it still does not appear (that I can see) anywhere in the app.

I'm just wondering as this Issue is still open, has this PR being merged with the master and is it also in the experimental branch(es)?  Also, should this functionality work for wifi wixels also?

If it's not there, can you point me to a branch that I can test with please?

Many thanks