StephenBlackWasAlreadyTaken/xDrip #63: Messed up bg data if sensor is started after insertion

raduiliescu, 2015-02-09:
Start sensor in app immediately after insertion;
After 2 hours double calibration was introduced.
Application displayed the all previous readings (for 2 hours) as 40 and further the algorithm was messed up.

Workarround was to stop and restart sensor.

Solution here seems to not add as BgReadings data provided in the first 1h 45mins from start sensor time.

StephenBlackWasAlreadyTaken, 2015-02-09:
Hey, there was actually a bug in how raw data was being time adjusted before vs after the initial calibration, this has since been fixed, you will want to make sure you pull in the latest commits!

raduiliescu, 2015-02-10:
I've gave it a try earlier with latest sources and seems to be working fine!

closed 2015-02-10 by raduiliescu