StephenBlackWasAlreadyTaken/xDrip #77: Alert on missing receiver connection

saercnap, 2015-02-25:
I've found the BLE module flashing due to not being connected, but Dexdrip is happily chugging along without data or notification

StephenBlackWasAlreadyTaken, 2015-02-25:
+1 on this, it should live in the dex collection service as thats something thats already restarting itself using an alarm system while its not connected, should be easy to implement

saercnap, 2015-02-26:
This should probably also/or alert on X minutes since last dexcom packet. eg, i'm paired to the HM-11 just fine, but the wixel isn't receiving packets

StephenBlackWasAlreadyTaken, 2015-02-26:
kk, since I dont want to wake the phone every minute to see if the last one was a certain number of minutes ago (for batteries sake)I think the best route would be..
When a packet is received, set a wake timer for say... 15 minutes from then (or x minutes, set in settings)
and if a packet is received less than x minutes from then, well change that wake timer to instead be x minutes from now.. and so on...
so the wake timer will only actually fire if nothing has been received (and thus moved the wake timer) for the full x minutes

do we continue to alarm every x minutes?

saercnap, 2015-02-26:
Sounds reasonable.

I suspect the continued alerting should be a preference. Chances are you either know it won't work (swimming?) or you don't (something is broken). Continued alerting for the former could be annoying if you don't want it.

StephenBlackWasAlreadyTaken, 2015-05-09:
currently in this pr: https://github.com/StephenBlackWasAlreadyTaken/xDrip-Experimental/pull/3

StephenBlackWasAlreadyTaken, 2015-07-24:
in beta now https://github.com/StephenBlackWasAlreadyTaken/xDrip/wiki/xDrip-Beta

closed 2015-07-24 by StephenBlackWasAlreadyTaken