StephenBlackWasAlreadyTaken/xDrip #84: Can we add Samsung Tizen support?

johnbarr00, 2015-03-17:
I recently purchased the Samsung Gear S smartwatch and I would love to see the NightWatch app be compatible with it. Is this possible? How complex is the process? I don't know a lot about coding and it took me a while to get my DexDrip unit up and running.

StephenBlackWasAlreadyTaken, 2015-03-19:
Their sdk does not look terrible, Im sure its quite possible, but I dont have a tizen device to write it for

johnbarr00, 2015-03-19:
I'd be willing to pay you for your time! I can test out whatever APK you send me. You can email me at johnbarr00@gmail.com if you want to talk more about it. 

Thanks again Stephen for all of this amazingness!

yazman697, 2015-07-15:
What was the answer to this?  I'd like to get my daughter a smartwatch so she doesn't have to carry a phone that could possibly be left somewhere or someone take it easily. The Gear S seems like the easiest route without having to order an unsupported network android watch from another country.

pgpalmer, 2015-07-15:
A couple of people are using ZGPAX S8 phones successfully.  It is an android watch from China.  I ordered a different model that wasn't good enough quality to use and sent it back, but am looking at the ZGPAX S8 and S7 models now.  They run android 4.4.2.  There are some other phones that look good, but run an older version of android.  Not sure how consistent the quality is though!

claclana, 2015-08-09:
Hello! 
have you any idea how the xDrip will work in the moto 360 ?
Thanks for all this sharing!!

StephenBlackWasAlreadyTaken, 2015-08-09:
Works fine for the 360!

claclana, 2015-08-09:
:)  :)  :)
Super!
you are awesome

Mondelman, 2015-12-21:
Do you know if you can use it directly with your watch or do you need to carry on your phone too?
Thanks!!

apabari, 2017-01-19:
Was Tizen support ever added?

sthurber79, 2017-09-08:
Any update on if Tizen support will possible be built in? Xdrip+ is the only thing keeping me with Android Wear.

trizt, 2017-12-29:
Sadly it's not just adding some code to xDrip that would make it run on Tizen, you need to write a whole new application. Sure you can certify and repack (including Android emulation libraries), but they do take a hefty price for it, if I remember it right it's a question about 5000+ USD. 

At the Tizen development portal they do have ideas how to run Android apps on Tizen, one is to install an Android emulator, this has the drawback that you need to start the Android emulator and then from there start the application.
Another method is similar to the Samsung solution, which will build you a C++ based wrapper around the Java and you need to include needed parts of the Android OS in the package too.

The best options IMHO is that someone downloads the Tizen SDK and starts porting xDrip and release the "tDrip" or what they want to call it.

vanelsberg, 2020-03-11:
See https://github.com/NightscoutFoundation/xDrip/issues/460#

**G-Watch version 1.2**

Brickmaster1, 2023-06-17:
I am looking for something like this that would be standalone. Everyone suggests G-Watch for tizen but I would like my gear sport to be able to connect directly to my dexcom. Is there still no way to do this?

> See [NightscoutFoundation/xDrip#460](https://github.com/NightscoutFoundation/xDrip/issues/460)
> 
> **G-Watch version 1.2**
 Speaking of which I'm just confirming, G-Watch does require a phone as it seems- right? Because if not I would happily use that.