StephenBlackWasAlreadyTaken/xDrip #86: Proper HM-10 Bluetooth module

rodrigobaquero, 2015-04-06:
Hi Stephen,

Let me congratulate you, fantastic job. I would like to make my own xDrip I live in Czech Republic and buying from US can cause some delays and complications. I have found a HM 10 Bluetooth but I'm not sure is the right one. Could you please advise me whether this is the right one to buy? Description and link to shop below.

HM-10 Bluetooth 4.0 RS232 UART Modul Master / Slave Arduino

Popis produktu

Bluetooth 4.0 modul podporuje master/slave
Bluetooth čip: CC2541
Bluetooth: V4.0 Ble
UART rozhraní
Provozní napětí: 3.3V
Rozměry desky: 27mm x 13mm

http://robotstore.cz/obchod/arduino/hm-10-bluetooth-4-0-rs232-uart-modul-master-slave-arduino/

Best regards,
Rodrigo

StephenBlackWasAlreadyTaken, 2015-04-06:
Yup, looks fine!
Could be tricky to solder but it should do the trick

rodrigobaquero, 2015-04-06:
Thanks so much for your response. I'll try it out I'm a Java developer I have some spare time on the weekends. I can have a look at the code and make my contribution :)

syntaxerr66, 2015-04-06:
IF you're going to go that route, go with an HM-11.  It's what replaced the HM-10, and just as tricky to solder :)

closed 2015-04-21 by StephenBlackWasAlreadyTaken