StephenBlackWasAlreadyTaken/xDrip #90: Can xDrip be used in Labview?

ijustlovemath, 2015-05-14:
My setup: Dexcom G4 platinum
My use case: using LV to create a glucose controller, I've done all the work of controlling, but still have to manually input values. Is there an API or something in xDrip that would allow me to pull individual values to be fed into my software?

Just found this awesome project, was starting to lose faith that I'd ever be able to close the loop.

StephenBlackWasAlreadyTaken, 2015-05-14:
I have no idea what labview is or how you get data into it, but xdrip does export its sqlite database, it also broadcasts data locally on android if you enable broadcast locally, and it sends data off using the standard nightscout rest api (and mongo). If you mean from the wixel itself, it sends data over usb (if it sees usb is attached) and uart at the same time on each gathered packet

ijustlovemath, 2015-05-14:
Okay, that's good to know. I'll be sure to post in this thread if I have any other questions on specifics. Thanks for doing this, it's amazing what you and Nightscout are doing for the diabetic community at large!

ijustlovemath, 2015-05-16:
Okay, another question for you. Which pieces of your code actually grab a value from the device? As in, if I compiled them and ran ./code, it would output the current value on the monitor. Also, can the Internet enabled device attached to the monitor be a laptop or does it have to be an android phone? I'm not too experienced in Java or Android stuff so I can't really tell where I should be looking. Thanks again!