StephenBlackWasAlreadyTaken/xDrip #92: BG Alert-Notifications show in snooze time.

AdrianLxM, 2015-05-21:
BG Alert-Notifications show, even if in snooze time. Sound doesn't play.

I guess in file https://github.com/StephenBlackWasAlreadyTaken/xDrip/blob/master/app/src/main/java/com/eveningoutpost/dexdrip/UtilityModels/Notifications.java in **line 214**
` } else if ((userNotification != null) && (userNotification.timestamp >= ((new Date().getTime()) - (60000 * bg_snooze)))) {`

the ` >= ` should be a ` <= `.

Greetings, Adrian

StephenBlackWasAlreadyTaken, 2015-05-21:
This one is actually intentional, I like having the alert showing threads whole time I am out of range but I don't want it making noise each time it updates, just while not snooze. This behavior is actually already changed though in the experimental repo (the entire bg alert system has been overhauled)

AdrianLxM, 2015-05-21:
Ok, thanks for the quick reply, I'll be waiting for the experimental becoming master then and just do a quick workaround for my own device. Great work you are doing.

closed 2015-06-02 by AdrianLxM