StephenBlackWasAlreadyTaken/xDrip #93: Battery voltage measuring circuit may waste >10% battery capacity

jamorham, 2015-05-26:
Hi, I was looking at the battery voltage measuring circuit and I think it can be improved. I modified it and it seems to work. Please find my working below and let me know what you think:

The voltage divider in the hardware_setup.md document uses a 1K and 2.2K ohm resistor. This creates a circuit with 3.2K ohm resistance which is always closed. 

At 3.7 volts this will use 1.15 milliamps constantly (3.7 / 3200). Over 48 hours (typical battery life?) this would be 55 mAh which is >10% of a 500mAh battery's capacity. 

Changing these resistors to both be 10K ohm gives a 20K circuit which would use 8.8 mAh over a 48 hour period which is ~1.6% of the battery capacity. 

My tests with these 10K resistors appear to work, with convincing millivolt readings reducing over time for the battery in a 48 hour period.

Depending on the current required/resistance of the ADC on the Wixel, it might be possible to increase these resistors further and get even better power reductions. I haven't investigated this yet. 

Changing the resistor ratios means that this voltage divider is now dividing the voltage in half. 

This reduces the accuracy of the original circuit, but has the advantage that batteries up to the 6.5volt the Wixel can support can now be used without overloading the ADC. 

It also slightly simplifies the voltage divider circuit, a builder doesn't have to worry about which resistor is which as they are both the same.

After re-soldering these resistors a few times myself, I can't imagine that anyone who has already built an xdrip is going to want to change their existing configuration! But I thought it was worth mentioning as an issue to improve the design and could be even more significant if people wanted to use even smaller batteries.

Please let me know any thoughts on this. I searched and I didn't find anything implying it had already been covered.

StephenBlackWasAlreadyTaken, 2015-05-28:
Good to know! I actually know very little about voltage and all that jazz so thank you! Ill definitely try this out!

syntaxerr66, 2015-05-28:
So, a couple of thoughts (and I'm the guy who built the circuit).  First, yes, the higher resistor values work better.  I was just going with what I had on hand at the time :).  The trade off to this is that the resistors don't read as rapidly, but I doubt it'd be noticeable for this circuit.  As for the ratios, LiPo batteries only go up to 4.2v for a single cell or multiple cells run in parallel.  You can't have multiple cells run in series because you'll hit max 8.4v with just two cells, which would kill the Wixel's VIN, and the ADC pins on the Wixel can't tolerate greater than 3v3.  

The reasons I picked the resistors I did is because of the common availability and the precision you can get.  So yes, a higher value can be used while still maintaining the same ratio, but there's no point to changing the ratio.

jamorham, 2015-05-29:
Okay good. I think whatever ratio or resistors used, it is good if the app allows for some way to calibrate for the battery so it's flexible whatever configuration people use. 

I have personally been working on an NiMH powered bridge for maximum battery safety and also a version to work with a gsm modem. My battery voltages have been up to 5 volts. I wanted to be super careful not to fry my wixel.

The current master branch of wixel-xdrip during my testing didn't work after the first sleep to read the battery level. I had to remove the makeAllOutputsLow(); and add setADCInputs(); after wake up. I am not sure that this is the correct way to solve the problem or if I'm using the most current software branch.

Cagier, 2015-07-24:
Can I ask a silly question please.  As the xDrip uses the HM10/HM11, why don't you just use the AT_BATC1 and AT+BATT commands to get the HM10 to measure the battery strength?  These are available from version v520 upwards.  My understanding is that this could be queried over UART by the Wixel and/or over BT by the Android app (using the AT command or the OnLeScan function result arrays).  After all, it is the same battery that is connected to the wixel and the HM10/HM11 (but maybe it is capped or regulated in some way before it gets there?)  No extra resistors or wiring involved!

If you were throwing in a few extra AT commands it would be also be sweet if the xDrip app also showed the HM10/HM11 firmware version and the IC Temperature as this information is also available using similar commands/functions.

Cagier, 2015-07-24:
Could I also suggest adding an optional setting in the xBridge2 app to start flashing the red LED on the Wixel when it detects that the battery is starting to get low.  This could increase with frequency/intensity as the voltage level drops.  I know that might ironically waste more battery but it might also prompt you to charge it quicker!

syntaxerr66, 2015-07-24:
The reason that reading the battery through the HM-1x wouldn't work is because the Wixel is putting out a straight 3v3 to the BLE module.  While it is the same battery, the voltage is being stepped down by the Wixel for the BLE module.  The module wouldn't be able to read anything other than 3v3 until the LiPo battery went below 3v3, so by that point you only have .3v to .5v left to measure before the battery dies.  The best you could get was a "Battery OK" and "Battery about to die" type message to the user.

Cagier, 2015-07-24:
Ah, that makes sense!  (See, I knew it was a silly question!)  Thanks syntax for the quick and comprehensive reply.  I might open a separate issue for retrieving the other HM10 info as it is not at all relevant in this context then.  Although my suggestion about using the led to reflect the battery state still stands!
Thanks again.

detroitdr, 2016-11-17:
Syntaxerr66, Thank-you for sharing, I'm currently working on one of these for my son. I can only find 5% resistors not 1%. Will that matter? Can i use the 5% resistors?

Thank-you in advance,