StephenBlackWasAlreadyTaken/xDrip #97: ZGPAX xDrip

yazman697, 2015-07-21:
I successfuly loaded the app on the device, but it won't allow me to scroll down to accept the End User agreement.  Is there a way to bypass this or edit a file on the device to show that I have accepted it?

StephenBlackWasAlreadyTaken, 2015-07-21:
You need to use the experimental branch of the app for now, those changes have yet to be moved over to the main repo

yazman697, 2015-07-21:
Pardon my ignorance, but where can I find that?

StephenBlackWasAlreadyTaken, 2015-07-21:
https://github.com/StephenBlackWasAlreadyTaken/xDrip-Experimental

There is no downloadable apk of it at the moment but I will be putting one up for the upcoming beta soon!

yazman697, 2015-07-21:
Please don't take this as pushy or demanding because I'm SO appreciative
for all the help you give,  but do you have an idea when that might be
available so I can get my daughter hooked up with her new cell watch?  Or
point me in the right direction to compile it myself? I tried throwing it
in the Android SDK and using eclipse to compile it but didn't have much
luck.  (I'm a cisco engineer,  not a programmer)

Thanks again for any assistance.

StephenBlackWasAlreadyTaken, 2015-07-21:
haha no worries, 
I generally recommend android studio over eclipse for android stuff these days, I think there is a nifty video by andrew on youtube (just search for xdrip or maybe dexdrip) but the beta itself should hopefully be up by tonight!

StephenBlackWasAlreadyTaken, 2015-07-24:
Beta is up https://github.com/StephenBlackWasAlreadyTaken/xDrip/wiki/xDrip-Beta

closed 2015-07-24 by StephenBlackWasAlreadyTaken