StephenBlackWasAlreadyTaken/xDrip #103: [Beta]

morganmott, 2015-08-04:
Unable to see data on Dexcom follow app.  I have Dexcom Share enabled in xdrip settings and have entered dexcom username and pw.  Any ideas?

StephenBlackWasAlreadyTaken, 2015-08-04:
and are you gathering data from a dexcom share itself?

morganmott, 2015-08-05:
No, not anymore.  I have changed the uploader phone from an iphone 5 to an
android.  I'm running the xdrip beta on the android uploader phone.  I've
got the xdrip settings for mongolab and dexcom share enabled.  I've entered
the dexcom username and pw that I had previously used on the share app we
were running on an iphone uploader phone.  The dexcom receiver is showing
the data and nightscout on my iphone displays current data.  I thought that
I could use both dexcom's follow app and nightscout at the same time on my
iphone to see data with the xdrip beta.  I liked the feature of  dexcom
follow of not having to have the app open for the alarms to go off  If
that's not possible, that's ok.  I'll get used to keeping nightscout open
all the time to receive alarms.  Thanks for your help.

oamyoamy, 2015-08-05:
I have also tried to use the upload to share feature since installing the beta 2.0 of xDrip and have not gotten it to work.

StephenBlackWasAlreadyTaken, 2015-08-05:
No sorry, I meant are you using a share reciever as opposed to the wixel. 

I'm hoping to roll out an update that will log the errors in a way that they will be easier for users to send me so we can see what's going on. 
Out of curiosity, is the share app still installed on the iPhone? Mind uninsalling it?

morganmott, 2015-08-05:
Yes, I'm using the dexcom share receiver.  Yes, the share app is still on
the iphone, I can uninstall it.

oamyoamy, 2015-08-05:
Also using the Share receiver, Stephen. I tried turning off my other upload (NS) to see if it would help. No luck.

oamyoamy, 2015-08-06:
Updated to the new version of the beta today -- Share server upload tests still not working.

StephenBlackWasAlreadyTaken, 2015-08-06:
Yup, nothing there has changed but it does now log the errors it hits, can
you check in settings, errors, and let me know what it says!

oamyoamy, 2015-08-06:
Under "low," a zillion of this error shows up:

BgReading
getForTimestamp No luck finding a BG timestamp match -- and date and time

Not sure if that's related to the Share issue or not. The only other few errors ("mid") are a cluster of missed readings, but those I knew about.

oamyoamy, 2015-08-06:
Oddly.... there are multiple line entries for this error per timestamp...
There are 12 of the errors for 4:41PM, 11 of them for 4:36PM, 10 of them for 4:31PM, etc.

StephenBlackWasAlreadyTaken, 2015-08-07:
Yeah that would be from each page it read from the dexcom I imagine (many
entries per page). But nothing about share in there? It would be under high

closed 2015-09-18 by StephenBlackWasAlreadyTaken