StephenBlackWasAlreadyTaken/xDrip #104: Delayed uploading

Alkene, 2015-08-10:
I'm running two uploading phones on two different dexcom shares and run into issues where the data hasn't been uploaded for 20 min or longer and its regularly in the 9-15 min since the last upload.  I'm not sure if the issue is between the dex reciever/app or the app/database.  Restarting the collector fix's it, but would be nice not to have to wake up and night and hit the restart collector to continue getting data when it goes past the 20 min and my alarm goes off.  Both phones are on wifi only and do not have active sim cards.  Any help would greatly be appriciated.

StephenBlackWasAlreadyTaken, 2015-08-10:
Are you running the beta currently?

Alkene, 2015-08-10:
No, I used the master branch on 7/22/15.  I can't seem to find a version anywhere.  Says version 1.0 in the app info in the phone settings.

StephenBlackWasAlreadyTaken, 2015-08-10:
Yeah, give the beta a shot and let me know if it works any better for you.
We added several wakelocks to help avoid the delayed issues!

Alkene, 2015-08-11:
K, will do.  Thanks,