StephenBlackWasAlreadyTaken/xDrip #106: [BETA] App does not stay connected with bluetooth

msinkovich, 2015-08-15:
As a heads up everything on 2.0.1 was working fine until I updated to the latest Google app (5.1.17.19).  Following that the bluetooth will not stay connected.  Im not 100% if this is the direct cause or if it is something else because I havent had the time to roll back a few things to make sure....

Im not even sure if the Google app would cause this but it was the only thing that changed on my phone yesterday.  I will continue to try to roll back to 5.1.17.16 or even roll back the Play Services but as of now the app will not stay connected to the receiver.

If there is something that I can get you to help let me know...

msinkovich, 2015-08-15:
in the logs I get this:
DexCollectionService
onServiceDiscovered: service 0000ffe0-0000-1000-8000-00805f9b34fb not found

msinkovich, 2015-08-17:
https://carousel.dropbox.com/photos/cc/uwjEXrznjZIIFBv

https://carousel.dropbox.com/photos/cc/dDlC8N1sFIUJEPR

msinkovich, 2015-08-17:
Ignore that last one... I forgot to change settings back

StephenBlackWasAlreadyTaken, 2015-08-18:
These are all due to lack of encryption, you will need to tell xdrip to forget the dexcom and vice versa as described here:  https://github.com/StephenBlackWasAlreadyTaken/xDrip/wiki/xDrip-with-Dexcom-Share#reconnecting-devices-if-they-are-unable-to-pair-properly

msinkovich, 2015-08-19:
Thanks.  I had only unpaired the device from the app not the receiver...

closed 2015-09-18 by StephenBlackWasAlreadyTaken