StephenBlackWasAlreadyTaken/xDrip #107: [BETA] No menu to export databae

AdrianLxM, 2015-08-15:
I'm writing in behalf of another user:

In the current beta 2.0.1 the menu button to export database (usually right upper corner) is no longer there.

[image]

The screen resolution is: 1920x1080

StephenBlackWasAlreadyTaken, 2015-08-18:
I thinkkkk this is just a samsung thing, It hides the menu and you have to use the 'menu' button on the phone instead

AdrianLxM, 2015-08-18:
Good to know. It indeed is a samsung galaxy note something.

I think we should include the export in the NavDrawer then. It would be good to have a predictable way to export the db when helping someone via chat/phone. The menu button may have different position and different icon on each phone.

StephenBlackWasAlreadyTaken, 2015-08-18:
I feel like I would rather it be in the settings as the current drawer is getting cluttered, thoughts?

AdrianLxM, 2015-08-18:
Settings sounds good to me. :)

closed 2016-01-20 by AdrianLxM