StephenBlackWasAlreadyTaken/xDrip #112: BLU Life Play Mini

jerrypk1, 2015-09-14:
Hi, we just received a dexcom share and were hoping to use the xDrip to connect to a recently purchased BLU Life Play Mini which has Bluetooth 4.0 and KitKat. When I start, it never seems to connect, although the status says connected. Is it possible to look at log messages or something to tell me what's wrong?  Thanks. Jerry

StephenBlackWasAlreadyTaken, 2015-09-14:
if you are running beta3 go into the app -> settings -> error log

jerrypk1, 2015-09-15:
I am using the built version on this site. I'm not sure if that's beta3 or not.  In any case, I don't see an 'error log' in the settings. Do you think this combo works - BLU Life Play Mini (Bluetooth 4.0 + Kitkat)?  I would think it should from what I understand.  Thanks

StephenBlackWasAlreadyTaken, 2015-09-15:
You can get the beta version here:
https://github.com/StephenBlackWasAlreadyTaken/xDrip/wiki/xDrip-Beta

jerrypk1, 2015-09-16:
Thanks...I'll try that tonight

jerrypk1, 2015-09-16:
The error log is very hard to read w/ yellow and orange background. I'm using Kitkat so I know I have to keep resetting both devices, etc. so I'm not always sure I've reset them properly.  In any case, I'm getting 'The phone is trying to read from paired device without encryption. Android Bug?' in the log.  I also see in the status 'BG Reading unavailable - xdrip data collection service is running.' Any ideas?