StephenBlackWasAlreadyTaken/xDrip #116: Mongo devicestatus field incorrect format

porkupan, 2015-10-01:
When using the Mongo API directly, the xDrip logs devicestatus entries with an empty "created_at" timestamp field.  This confuses the Nightscout site when switching from Mongo API to REST API.  The REST API always fills out the "created_at" field.  When obtaining the current battery status, the Nightscout treats the last entry with empty timestamp as the most recent (probably a bug in Nightscout code).  

xDrip can avoid this confusion by always filling out the timestamp.

StephenBlackWasAlreadyTaken, 2015-10-01:
Interesting! This is super simple to add, Ill throw a pr up in a few and ping you on it if you are willing to test!

porkupan, 2015-10-02:
Sure.  But it's very easy to test, just switch from Mongo to REST API, and watch the battery status stop updating.  Then delete all the entries created via Mongo API, and watch the battery status show correctly.  As long as there are any devicestatus entries with empty "created_at" field in the table, the Nightscout is going to "ignore" entries with non-empty "created_at".