StephenBlackWasAlreadyTaken/xDrip #121: xDrip on Android 6 cannot find Bluetooth devices.

yoda226, 2015-11-20:
I have used xDrip with my Nexus 5 with Lollipop and Dexcom Share for months, but after upgrading to Android 6, xDrip can no longer find the Dexcom Share when scanning for Bluetooth devices. In fact, it does not seem to find any Bluetooth devices. I have tried various different things to try and get it to work without any luck. I have even tried updating xDrip to the 2.0.4 beta version, hoping that might fix it, but, again, no luck. Any help that you could offer would be greatly appreciated.

rpzigler, 2015-11-22:
I also am having this problem. I recently upgraded to a Nexus 6P and since then I have been unable to get xDrip to setup correctly. I'm guessing the major changes with Doze are making the current app and beta incompatible with Marshmallow. Hopefully this will be figured out soon. I miss being able to see everything on my phone! :(

AdrianLxM, 2015-12-11:
The current master branch of xDrip-Experimental already has the marshmallow capabilities.
You may want to try it.

scottc5, 2015-12-12:
I couldnt get xdrip or xdrip experimental to work until I added these changes in to both, then they both started working on my Nexus 6p... https://github.com/StephenBlackWasAlreadyTaken/xDrip/pull/122

AdrianLxM, 2015-12-12:
@scottc5, If you already have them edited in the xDrip-Experimental codebase, could you please send a PR?

dirtydave0221, 2016-03-17:
Just tried pairing with my Nexus 6p running Chroma (rooted) and it does not show up as a bluetooth device.  I can potentially help out with debugging as needed.

StephenBlackWasAlreadyTaken, 2016-03-17:
You probably need to allow it permissions, go into your phone settings,
apps, xdrip, permissions and make sure it has the proper permissions, let
me know if still no dice

dirtydave0221, 2016-03-17:
Still no cide on that @StephenBlackWasAlreadyTaken.  I'm looking to switch roms this weekend, but can do some exploratory things tonight if that helps you out

aaroecker, 2016-03-18:
@stephenblackwasalreadytaken this has been an on going issue for me too; a very frustrating one at that.  2.0.3 seems to be the most stable, but I often need to hit the restart collector button if I leave Bluetooth range and come back.  Bluetooth reconnects with 2.03, but the data doesn't get uploaded, it's odd.  On the latest release of xDrip the phone would need to be restarted to get a good connection and even then the Bluetooth would disconnect randomly even when in range.  The error log says that it's a status bug 133 causing the issue and I need to forgot the devices and reconnect.  The issue wasn't present with my Nexus 6 running marshmallow, but is present with my Moto X Pure running marshmallow.  Any insight would be greatly appreciated.  I'm currently using 2.03.

dirtydave0221, 2016-03-18:
@aaroecker I've been experiencing similar.  I moved to Pure Nexus rather than Chroma (I needed to update) and now things were able to sync up, but when I connected to my car's bluetooth and then disconnected, the data didn't update until I restarted my phone.

aaroecker, 2016-03-18:
@dirtydave0221 I haven't rooted and romed, my phones for sometime.  Android has gotten so good that I haven't felt the need to.  The thought has crossed my mind though to troubleshoot this issue.  For heaven's sake I even contemplated jumping ship and moving to iOS 😵

Cagier, 2016-06-13:
Just because the O.P. and others reported not being able to find the xDrip/xBridge/DexcomShare as a bluetooth device and others may stumble across this thread, I thought I should mention that I had a similar issue in issue #137 and my specific  problem was also related to Android 6.  I had previously looked at Location permissions and didn't see anything obvious so dismissed this but ultimately this was in fact the issue for me.

Everything had worked fine until the xDrip lost the connection and on an attempt to forget and then re-connect I lost the ability to successfully scan for BT devices in xDrip.  The solution was to upgrade to the latest xDrip experimental version and edit the location permissions in Apps/Settings(cogwheel)/App Permissions/Location/xDrip.  Most of you are probably already aware of this fix/workaround but no harm to include it here in the hopes that it might save some other poor soul from hours of resets, reinstalls and troubleshooting....

bigtunacan, 2016-09-27:
Not sure why it would impact this, but enabling location permissions fixed this for me as well.