StephenBlackWasAlreadyTaken/xDrip #123: Voltage measuring circuit

starkjoh, 2015-11-26:
Good afternoon,

I am a newbie in this project and I have very little experience with electricity in general. I startet to built my own xDrip-wixel about 2 weeks ago and it works very well. This little thing in my pocket is just amazing.
First, I did not use the voltage measuring circuit at all but it I realized it is a nice feature to have when I woke up in the morning and did not see any BGs. So I added 2x 10k resistors to the wixel. However, I could not find out if the battery life is somewhere shown in the xDrip app on the phone.
Then I found out about xBridge2, which actually has that feature enabled. But I have problems with BG gaps, which I think is related to the wixel quality. The normal xDrip circuit worked a lot better for me and I will switch back to it again. Now I am a little bit confused because there are resistors (1k & 2.2k) built in in the hardware_setup.md for the normal xDrip. I just don't get why they are used since battery life is not displayed in the app with hardware data source set to 'Bluetooth wixel'. Or is it anyhow possible? Am I missing something? 

Thanks a lot in advance!
Best regards from germany
Johannes

JohanDegraeve, 2015-11-26:
Hi,

i installed this on the wixel : https://raw.githubusercontent.com/jstevensog/wixel-sdk/master/apps/xBridge2/xBridge2.wxl
(just download and install with wixel configuratin utility) - i assume you did the same ?

Then if you use the latest beta version https://github.com/StephenBlackWasAlreadyTaken/xDrip/wiki/xDrip-Beta
you should see the xdrip battery level. Actually it's there in Beta 2.0.0

Best regards from Belgium :)

Johan

starkjoh, 2015-11-27:
Hi Johan,

thanks for the quick response!
That is exactly what I did. My question was if it's possible to see the battery level with xdrip (used to be called dexdrip
https://github.com/StephenBlackWasAlreadyTaken/wixel-xDrip/tree/master/apps/dexdrip), 
and not with xbridge2 (https://github.com/jstevensog/wixel-sdk/tree/master/apps/xBridge2) 
because xbridge2 uses another way to communicate with the phone. I had some problems there but they seem do be gone after I switched the power source from usb to normal battery.
But thank you anyway for your answer! :)

Regards
Johannes

closed 2015-11-27 by starkjoh