StephenBlackWasAlreadyTaken/xDrip #127: Bluetooth - asking for a PIN

exceter, 2016-01-03:
[image]

Hi,
yesterday finally i connected Wixel with BT. but there is a problem - it is asking for a PIN code. Could not find, i do not think it must be actually. nowhere it was mentioned.

[image]

Tried with:
i9250 (Android CM 5.1.1, BLE supported)
A21+ (Android 4.2)
another chinese phone with Android 4.4 does not see the device

exceter, 2016-01-03:
Actually sensor is not installed yet, does that make sense?
As i understand transmitter does not stop working

xitation, 2016-01-03:
Often the default pin for the HM10 and HM11 based chips is "000000" without the "".
You would find this pin the the documentation for the HM10/HM11 chips, but it is rather hidden and hard to find from memory. 

Chances are you can't see the device on your other phone as it might not support Bluetooth low energy mode which is what the HM10/HM11 devices use.

exceter, 2016-01-03:
xitation,

thank you very much, '000000' worked.

amlynek, 2017-06-26:
Don't pair in your phone's BT settings. Scan and click on the xBridge device in the xDrip app. No password needed.