StephenBlackWasAlreadyTaken/xDrip #128: Dexcom G5 Support

kskandis, 2016-01-05:
Does (or will) xDrip support the latest Dexcom G5? I am currently using xDrip with Dexcom G4 Share but will be upgrading to G5 when my current transmitter expires.  I'd like to continue to use xDrip rather than Dexcom's iPhone sw since I am using an Android phone (which Dexcom G5 does not support).  Plus, it appears the G5 is not integrated with Dexcom Studio and the actual cgm numbers are not available via Dexcom Clarity export.  I also use Nightscout with your xDrip which I believe is real-time vs Dexcom's Clarity server that has 3 hour delay (I believe)!

oglematt, 2016-01-07:
I was also wondering about this. My daughter is currently using the G4, but we'd love to go to the G5 at some point to remove a device from the mix. I hate that she has to carry around so much stuff!

I really appreciate what you are doing here. It's made a huge impact on my family. Is there somewhere we can donate to say thanks?

johnipayne, 2016-01-16:
I am interested in this as well.  Just got my G5.  If there is already a software dev working on this I would like to talk with them.  I would be happy to contribute code towards this end.

msinkovich, 2016-02-20:
Any more movement on development for this?

johnipayne, 2016-02-20:
I have a busy work schedule right now, but am planning on starting work on this when my schedule clears up this summer (probably July).

chrisharris751, 2016-03-11:
I have a little coding experience and a G5. If there is anything I can do to help, I'd like to.

TecMunky, 2016-05-13:
I am getting a G5 next week and am very interested in this project.  Is there any more news about using xDrip with the G5?

StephenBlackWasAlreadyTaken, 2016-05-13:
There are still some kinks to be worked out but things are working out
pretty well for a majority of users.

psjames, 2016-05-13:
Is it Beta 5 that works with the G5 or is there a different location to download a version that works with the G5?

rizzjc, 2016-06-03:
I built the app and installed on a Galaxy s4 with lollipop, but it doesn't connect to the transmitter.

StephenBrown2, 2016-06-04:
Built and installed Experimental 2.0.6_1 (5b6c42f) and installed on a Nexus 6P running N.
Gave it Camera, Location, and Storage permissions.
Can't connect to my G5 transmitter either.
Are there logs or a debug report I can give to help?

rizzjc, 2016-06-04:
I tried and tried, but my S4 didn't even see the bluetooth device. I tried rebooting, turning bluetooth on and off, etc. Then in the morning when I connected my phone to my car bluetooth, I noticed that sometime during the night it had paired with the G5, so I'm optimistic. Now I need to wait until the person it is attached to comes back home so I can try again.