StephenBlackWasAlreadyTaken/xDrip #129: xDrip app do not picking up data from BleMini

michalkominek, 2016-01-11:
Hello to all.
I have built my xDrip for my 5years old boy.
xDrip is based on:
1.	Wixel CC2511F32
2.	Ble Mini from Red Bear (http://redbearlab.com/blemini/)
3.	LiPoly charger
4.	550mAh LiPoly Battery
Wixel and wixel software is working properly (receiving data from transmitter and yellow LED is indicating correctly as in description).
BLE module is also visible for my phone (Samsung GrandPrime SM-531F)

My xDrip app is not receiving correct data from BLE Mini.
It shows “Please wait, need 2 readings from transmitter first."

BLE Mini default setting are:
 UART - Baud rate is 57600, 8 data bits, no parity, one stop bit & no flow control
So I wrote:

void initUart1() {
    uart1Init();
    uart1SetBaudRate(57600);

in my xDrip code for wixel.

After those settings it looks like my transmission is working because I receive some data on additionally installed Ble Mini app.

[image]

Can you answer what kind of data I should receive from my BleMini.
Are they correct? 
What all those numbers mean?
Is something missing?

Thanx in advance.

AdrianLxM, 2016-01-12:
Hi @michalkominek,

what you are seeing is the unfiltered raw data and then the battery level of the transmitter (217).
217 is pretty much full; 207 is the level the dexcom receiver will alarm that the battery is empty.
This is how xDrip parses this data: https://github.com/StephenBlackWasAlreadyTaken/xDrip-Experimental/blob/master/app/src/main/java/com/eveningoutpost/dexdrip/Models/TransmitterData.java#L57

I hope I could help you.
The development is done here: https://github.com/StephenBlackWasAlreadyTaken/xDrip-Experimental
And the developers chat is here: https://gitter.im/nightscout/dexdrip
Feel free to join :)

JohanDegraeve, 2016-01-12:
@AdrianLxM sure dex gives the alarm at 207 ? i think it's 210 that the alarm is generated.
207 means "empty", not sure if it can ever reach that level

AdrianLxM, 2016-01-12:
@JohanDegraeve I've seen values less than 200 that still worked. Once it reaches 207 the decay is pretty fast. And I'm pretty sure about the 207. xDrip alarms at 210 ("LOW") though and at 207 will show "Empty".

JohanDegraeve, 2016-01-12:
hmm, got an alarm now at 210, transmitter is 1 year old
already bought a new on but I might not need it in the next months ?

AdrianLxM, 2016-01-12:
I have no experience with the slim transmitter yet, but from 210 to 207 it can be a month or just a week with the other one.

michalkominek, 2016-01-12:
Thanks guys but my problem is so xDrip app is not picking this data up.
xDrip app shows “Please wait, need 2 readings from transmitter first." for hours...

About data format
Shouldn’t be space between 123 456 789 0 instead 123456 789 0 ?

michalkominek, 2016-01-12:
For me first 3 digits are actual BG 112, 118 ,127... etc. I did compare them with dexcom.

AdrianLxM, 2016-01-13:
This is coincidence. But that the raw data ofteb is not that far away from what the calibrated result will be speaks for the quality of the dexcom sensors.

michalkominek, 2016-01-13:
So you mean data format is OK?
If yes...Why my xDrip app don’t pick them up?
Can be because of some settings? RS or BT?

Cagier, 2016-01-13:
Can I ask a really silly question?  Did you actually try entering in 2 calibration entries or did it now allow you?

2 hours after starting a new session, you need to enter in a "double calibration".  If the session is already over two hours then you would be prompted immediately.  However, the xDrip app needs to receive one or two readings from the xDrip device first (i.e. the phone needs to get a couple of readings over bluetooth before you can calibrate).

Just from the description of your problem I want to ensure that you;
1) Started the sensor session (sounds like you did as you got a prompt to say that you will need to enter some values)
2) That the session is over 2 hours old (you can set this to be an older date/time when you do session start if that is already the case or you can make it older anyway just to test.)
3) That you have received at least 2 readings (just leave the everything beside your son for 10-15 minutes)
4) That you then selected the menu option to enter a (double) calibration.  (This will only appear on the menu once the above steps have been followed.)

Hopefully that is not patronising but I am just going to back to first principles on your query as you seem to be receiving data to see if there is anything straightforward that might get this working for you...

Cagier, 2016-01-13:
Sorry, re-reading above and it does seem that the two hours has passed and it is waiting for data before it prompts for the calibration.

A few other things that you might check:
1) You should not pair your phone over bluetooth with the xDrip bridge.  It should only be scanned for under Bluetooth devices on the xDrip app.  If you did pair then maybe try "forgetting" it in your Bluetooth settings. (Otherwise you may have problems in the app but you will also have poor battery performance on your phone.)

2) It is possible that the BLE Mini app is grabbing the packets first and so the xDrip is not getting them.  I would try and remove/forget the xDrip bridge from the BLE Mini app (or fully close/terminate/end the app before starting the xDrip app).

If you can see the xDrip on the Bluetooth scan within the xDrip app then you should be OK.  However, once it is selected it will then disappear but you can check that it is CONNECTED under the System Status Menu.  Can you confirm if you are connected using that menu?

michalkominek, 2016-01-13:
•	BLE Mini app is off
•	I’m connected according to Status menu.
•	It is not possible to set calibration values because this menu is not available
•	It is possible (theoretically) to stop and start the sensor
•	After 2h I get message “Please wait, need 2 readings from transmitter first."
This message stays for hours.

I suspect some setting issues from BLE mini (RTS/CTS flow?) but why I can receive data on BLE mini app?

Cagier, 2016-01-13:
That does seem bizarre because you are connected to the device so it can see the output from the Bluetooth and the output from the Bluetooth observed by BLE mini shows that it is picking up valid values and sending them over the air.  The three values are the raw data, dexcom transmitter battery and (optional) xbridge battery and they all appear to be valid values.  I think, on that basis, that your RTS/CTS flow is fine...

The only apparent change is that you changed the baud rate to 57600.  However, I would think that this should not matter if the xDrip app is saying that it is connected.  The data appears to get from the wixel out over the air on the Bluetooth.  I'm not so hot on the Bluetooth side so I don't know if there are any other Bluetooth settings that might need to change as the BLE Mini is not the standard HM11/HM11 module that is normally used for the bridge.  It looks like the data is being sent over Bluetooth and the module is connected but the xDrip cannot receive or understand what it is seeing.  (But I am only speculating going on the information above.)

That is pretty frustrating alright!  Sorry that I am not helping much here.  My last suggestion is to go into the settings menu and select "View Recent Errors/Warnings".  There may be some Bluetooth related warnings there?

Ideally you would try a different Bluetooth module but that will require major surgery.  Although it shouldn't make any difference (but as it uses a different underlying protocol over Bluetooth) it might also be worth considering using the xbridge2 app (in https://github.com/jstevensog/wixel-sdk ) and seeing if you have any success with that.  The xDrip app may just listen to it in a slightly different way but I am just grasping at straws here.

Check the error messages anyway and see if that provides any leads...

StephenBlackWasAlreadyTaken, 2016-01-13:
Can you double check the service and characteristic UUids, I believe they are different between the hm10/11 and the redbearlab devices.

I think I checked them at one point and documented them here https://github.com/StephenBlackWasAlreadyTaken/xDrip-Experimental/blob/master/app/src/main/java/com/eveningoutpost/dexdrip/UtilityModels/RedBearLabAttributes.java

but make sure yours are the same.

I dont believe I ever put the code up to make the redbearlab mini work, but it should be a minimal amount of work

michalkominek, 2016-01-13:
Thanx for suggestion.
I think exactly the same (xDrip app is not able to read received values for some reason)
I will try with recommended HM10 or HM 11 
According to your recommendation it should be CC2540?
By the way
Where can I find “View Recent Errors/Warnings?

Cagier, 2016-01-13:
Stephen is The Guru and he has spoken.  So, it does look like the xDrip is not expecting to see the Red Bear module and so is not communicating correctly with it.  However, if you look at the code he has pointed to then you should hopefully be able to get this working.  (Although you may need to compile your own version of the xDrip app.)  As he says, check your service and characteristic UUids.

If you have a HM10 or HM11 then that should work but you will have to rebuild your bridge which is a shame.

To answer your previous question anyway, that error log option is under the "Settings" menu (at the bottom).

Let us know how you get on.  Good luck!

michalkominek, 2016-01-13:
Thank you Guru
Thank you Cagier

To be honest I have no skills to program Android app.
So I have to handle that shame :-( and rebuild my bridge.
I will let you know about results.

StephenBlackWasAlreadyTaken, 2016-01-13:
hahaha cc2541s are fine as well, most hm10s ship with them currently, there was a product called a cc41 that was kind of a knock off hm10 but it had a different descriptor uuid than the app was expecting so I warned against using it.

The code has since been updated so as long as you dont get a dud, even the cc41 will work just fine.

The amount of work to get the different device recognized should be minimal, I cant do it myself at the moment but it would probably be much less work than resoldering everything.

michalkominek, 2016-01-13:
Resoldering is my plan for Friday evening with bottle of "RedBear" :-)

michalkominek, 2016-01-21:
My xDrip based on HM-10 is finally working without any issues.
Thank you for support.

closed 2016-01-21 by michalkominek

detroitdr, 2016-11-22:
Hello all,
I'm trying to get the xdrip wireless bridge to work for my son, but I am unable to communicate to the HM10.

HM10 appears to be real, according to the picture layout of the chips and it's a CC2541
I have wired two bridges, the old way and new xbridge2 way. Both have a problem connecting to the HM10 for me to update the firmware to 541.

I'm not a programmer so i've been using the instructions here: https://github.com/jstevensog/wixel-sdk/tree/master/apps/xBridge2

I have loaded the wixel with the USB_serial.wxl, and grabbed the com port (6)
I closed the wixel configuration program on (Win 10) and loaded up Putty (also tried Hype!Terminal) with 
9600,8, parity NONE, Bits:1 Flow Control none.
The wixel yellow light turns on. I send an AT only and the green light flashes quickly but text received is nothing on the terminal.
BAUD rate: I have tried EVERY baud rate but that doesn't help either.

I built two of these (old and new way). I find it hard to believe i've screwed both up. I've checked wiring and they appear fine.  Below is a link to my rigs

Again i'm not a programmer and i've been re-reading all the instructions on many different pages but I'm stuck. I'm hoping someone says ahhhh you needed to do this.....

Any help please would be greatly appreciated. As parents, we don't get much sleep and hoping this will let us get some...
Tired but excited to finish this xdrip,
George

[image]