StephenBlackWasAlreadyTaken/xDrip #130: Android alarms set by xDrip since 2.0.5

starkjoh, 2016-01-12:
Hello all,

the following issue occured after I updated to v2.0.5:
The Android alarm symbol started to show up continuously in the status bar and there are android alarms set about every 5 minutes, sometimes the time period varies. For example, when it is 10:01 I can see that an alarm is set to 10:06. When it passes 10:06 the phone does not ring or anything, it will just set another alarm to 10:11 and so on. 
As a result the alarm symbol in the status bar won't go away. That problem does not occur in version 2.0.3.
I am running android 5.1 (HTC One M8).

Thanks in advance,
Johannes

whoiswes, 2016-01-13:
This is probably due to the MM changes AdrianXL pushed, as I have a similar issue with Nightwatch master from the middle of December (on Android 6.0.1).

starkjoh, 2016-01-13:
Okay, then there won't be a fix for this i guess.. Thanks for the info though.

aaroecker, 2016-01-18:
I'm seeing this issue too.  Will there be a fix?

StephenBlackWasAlreadyTaken, 2016-01-18:
You may need to head into your phone settings -> apps -> xDrip -> Permissions and enable location or whatever else it asks for (Maybe? or maybe thats just for M)

aaroecker, 2016-01-18:
That didn't seem to fix it.  On another note, do you recommend xDrip to be battery optimized or not in Android 6?

AdrianLxM, 2016-01-18:
@aaroecker, some parts won't work if it is battery optimized. When xDrip is first started it will ask to disable battery optimization.

aaroecker, 2016-01-20:
This morning I did a clean install of xDrip after a reboot and clearing the cache in the system recovery.  I turned off battery optimization and allowed for all permissions before opening and setting up the app.  The alarm bug is still present.  Just an fyi.

AdrianLxM, 2016-01-20:
So far I don't see any bug. It is just how Android notifies the user that a certain kind of (reliable) timers are set. We need to use them now as some of the battery optimizations in Android 6 would lead to delayed alarms otherwise.

Stephen's response to the permissions was regard to this issue I suppose: https://github.com/StephenBlackWasAlreadyTaken/xDrip/issues/132

aaroecker, 2016-01-20:
Ok, thanks for clearing that up.

grincher, 2016-02-07:
I'm seeing the every 5 minutes alarm issue.  xDrip 2.0.5 beta on a Moto X (2nd gen) Android 6.0.

AdrianLxM, 2016-02-07:
@grincher What alarm issue? Do the alarms play/vibrate every 5 minutes even though you have snoozed them for a longer period of time?

grincher, 2016-02-07:
sorry I didn't explain in more detail.  when you pull down the
notifications from the top of the display, Android reports there is an
alarm set in 5 minutes.  then when that time arrives a new alarm time 5
minutes out is again set.  I assume this is tied to how xDrip wakes to
perform its update.  so, there is no alarm heard, but it defeats the
Android feature trying to tell the user about the next user-set alarm.
perhaps there is a setting to tell Android not to advertise this alarm.
thanks!

AdrianLxM, 2016-02-07:
@grincher Sorry, I could have looked at the thread you were posting in.
As of the documentation it is not possible - or in fact it is possible but the system can then delay the call for up to 15 minutes. (It allows just one call every 15 minutes.)
Having alarms like the "Missed Reading Alert" delayed by another 15 minutes does not seem good to me so I'd rather go with the icon. And "Missed Readings" usually implies multiple timers (one for the alarm, the other before that to try and reconnect the bluetooth connection) so it seems quite likely to fall into that trap. :/

grincher, 2016-02-07:
this is definitely an area where I am not an expert (I do embedded software).  perhaps Android doesn't provide a way to accomplish this here because it assumes this type of code would exist in a "Service" instead of regular app.  there certainly should be an api somewhere to be able to have these things triggered timely but not revealing evidence of internals to the user.  regarding the icon at the top left of the display--I don't mind this at all.  thanks for the quick response!

starkjoh, 2016-02-10:
Thanks for all the information! I think it's all clear now.

Artorias360, 2016-02-11:
so, there is no solution? the alarm icon will be always on? It is very annoying

starkjoh, 2016-02-15:
I just removed the the complete alarm icon from the status bar. It's less annyoing now :P

closed 2016-02-25 by starkjoh

hakkalugi, 2016-08-17:
@starkjoh May I ask how exactly you disabled/removed the alarm icon from the status bar?
Are you on Marshmallow? Are you rooted? -- Unfortunately I'm on Lollipop and can't seem to find a way. Although I'm tempted to just write my own simple app to help further clean my status bar.

My personal opinion is that I know the alarms I've set on my device, I don't need a reminder of various active settings.

Thanks

starkjoh, 2016-08-18:
I used the Android System UI Tuner. I'm not sure but I think this feature is only available on Marshmallow devices. 
Here is a how to: [http://android.wonderhowto.com/how-to/enable-hidden-system-ui-tuner-menu-android-marshmallow-0165141/](System UI Tuner)

hakkalugi, 2016-08-19:
Yup, System UI Tuner is unfortunately only Marshmallow and greater.

Thanks for the response!