StephenBlackWasAlreadyTaken/xDrip #131: [Beta5] Upload to Mongo / Nightscout is stuck

jmercay, 2016-01-15:
I've tried both direct Mongo upload and REST upload. After a few minutes, no more data is uploaded to Nightscout, and I see the following in the logs:

MongoSendTask wakelock acquired NN

with NN incrementing, and no corresponding wakelock released message.  

Killing xDrip (Apps->xDrip->Force Stop) resumes the process, but the uploader get stuck again shortly. Happy to share more about my config if it helps.

AdrianLxM, 2016-01-16:
@tzachi-dar, can you help with that?

tzachi-dar, 2016-01-16:
Hi, 
jmercay has the uploading finished after some time?

After killing, and trying again, did the wakelock start from 0 or did it start from bigger number again.

When it was first stuck, what method have you been using? mongo or rest?

Were you using xDripviewer mode? (If you don't know what that is) than I guess that you were using the default.

Thanks
Tzachi

jmercay, 2016-01-23:
Hi,

No, the upload is clearly stuck for hours. When I kill the app, the wakelock counter restarts from 0 and the past data gets uploaded almost immediately.

I've been switching between mongo and rest. First mongo, then rest when the problem started, and now back to mongo.

I'm not using xDripViewer: a single phone with xDrip is getting data from Dexcom Share and uploading to mongo. A couple devices sync down the data with Nightwatch.

tzachi-dar, 2016-01-23:
Hi,

Can you connect to your phone and run adb commands?

When the uploader is stack, I need you to run the following commands:

1) adb shell ps | grep -i xDrip
This will give you the process number.

2) adb shell kill -3 process_number
3) cat /data/anr/traces.txt

This should give us the call stack of where xDrip is stuck.

Thanks
Tzachi

jmercay, 2016-01-26:
I haven't got a userdebug phone (so no root and kill -3 doens't work). But traces.txt contains the following:

"main" prio=5 tid=1 TimedWaiting
  | group="main" sCount=1 dsCount=0 obj=0x7506f2a0 self=0xb877c8c0
  | sysTid=386 nice=0 cgrp=default sched=0/0 handle=0xb6f9db34
  | state=S schedstat=( 0 0 0 ) utm=2391 stm=72 core=3 HZ=100
  | stack=0xbe080000-0xbe082000 stackSize=8MB
  | held mutexes=
  at java.lang.Object.wait!(Native method)
  - waiting on <0x014fb988> (a java.lang.Object)
  at java.lang.Thread.parkFor$(Thread.java:1220)
  - locked <0x014fb988> (a java.lang.Object)
  at sun.misc.Unsafe.park(Unsafe.java:299)
  at java.util.concurrent.locks.LockSupport.parkNanos(LockSupport.java:198)
  at android.database.sqlite.SQLiteConnectionPool.waitForConnection(SQLiteConnectionPool.java:670)
  at android.database.sqlite.SQLiteConnectionPool.acquireConnection(SQLiteConnectionPool.java:348)
  at android.database.sqlite.SQLiteSession.acquireConnection(SQLiteSession.java:894)
  at android.database.sqlite.SQLiteSession.executeForCursorWindow(SQLiteSession.java:834)
  at android.database.sqlite.SQLiteQuery.fillWindow(SQLiteQuery.java:62)
  at android.database.sqlite.SQLiteCursor.fillWindow(SQLiteCursor.java:143)
  at android.database.sqlite.SQLiteCursor.getCount(SQLiteCursor.java:132)
  at android.database.AbstractCursor.moveToPosition(AbstractCursor.java:219)
  at android.database.AbstractCursor.moveToFirst(AbstractCursor.java:258)
  at com.activeandroid.util.SQLiteUtils.processCursor(unavailable:-1)
  at com.activeandroid.util.SQLiteUtils.rawQuery(unavailable:-1)
  at com.activeandroid.util.SQLiteUtils.rawQuerySingle(unavailable:-1)
  at com.activeandroid.query.From.executeSingle(unavailable:-1)
  at com.eveningoutpost.dexdrip.Models.Sensor.currentSensor(Sensor.java:67)
  at com.eveningoutpost.dexdrip.Models.BgReading.latest(BgReading.java:485)
  at com.eveningoutpost.dexdrip.Models.BgReading.currentSlope(BgReading.java:186)
  at com.eveningoutpost.dexdrip.UtilityModels.BgGraphBuilder.unitizedDeltaString(BgGraphBuilder.java:367)
  at com.eveningoutpost.dexdrip.wearintegration.WatchUpdaterService.dataMap(WatchUpdaterService.java:181)
  at com.eveningoutpost.dexdrip.wearintegration.WatchUpdaterService.resendData(WatchUpdaterService.java:149)
  at com.eveningoutpost.dexdrip.wearintegration.WatchUpdaterService.onStartCommand(WatchUpdaterService.java:98)
  at android.app.ActivityThread.handleServiceArgs(ActivityThread.java:3028)
  at android.app.ActivityThread.-wrap17(ActivityThread.java:-1)
  at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1452)
  at android.os.Handler.dispatchMessage(Handler.java:102)
  at android.os.Looper.loop(Looper.java:148)
  at android.app.ActivityThread.main(ActivityThread.java:5443)
  at java.lang.reflect.Method.invoke!(Native method)
  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:728)
  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:618)

"Binder_3" prio=5 tid=17 Native
  | group="main" sCount=1 dsCount=0 obj=0x133170a0 self=0xb88d61f0
  | sysTid=412 nice=0 cgrp=default sched=0/0 handle=0xa1976930
  | state=R schedstat=( 0 0 0 ) utm=1891 stm=16 core=2 HZ=100
  | stack=0xa187a000-0xa187c000 stackSize=1014KB
  | held mutexes=
  kernel: (couldn't read /proc/self/task/412/stack)
  native: #00 pc 0002eb80  /system/lib/libsqlite.so (???)
  native: #01 pc 0000ffc5  /system/lib/libsqlite.so (sqlite3_step+724)
  native: #02 pc 000755f1  /system/lib/libandroid_runtime.so (???)
  native: #03 pc 00a1a187  /system/framework/arm/boot.oat (Java_android_database_sqlite_SQLiteConnection_nativeExecuteFo
rCursorWindow__JJJIIZ+146)
  at android.database.sqlite.SQLiteConnection.nativeExecuteForCursorWindow(Native method)
  at android.database.sqlite.SQLiteConnection.executeForCursorWindow(SQLiteConnection.java:843)
  at android.database.sqlite.SQLiteSession.executeForCursorWindow(SQLiteSession.java:836)
  at android.database.sqlite.SQLiteQuery.fillWindow(SQLiteQuery.java:62)
  at android.database.sqlite.SQLiteCursor.fillWindow(SQLiteCursor.java:143)
  at android.database.sqlite.SQLiteCursor.getCount(SQLiteCursor.java:132)
  at android.database.AbstractCursor.moveToPosition(AbstractCursor.java:219)
  at android.database.AbstractCursor.moveToFirst(AbstractCursor.java:258)
  at com.activeandroid.util.SQLiteUtils.processCursor(unavailable:-1)
  at com.activeandroid.util.SQLiteUtils.rawQuery(unavailable:-1)
  at com.activeandroid.query.From.execute(unavailable:-1)
  at com.eveningoutpost.dexdrip.Models.BgReading.latest(BgReading.java:487)
  at com.eveningoutpost.dexdrip.UtilityModels.BgGraphBuilder.unitizedDeltaString(BgGraphBuilder.java:361)
  at com.eveningoutpost.dexdrip.wearintegration.WatchUpdaterService.dataMap(WatchUpdaterService.java:181)
[... 19 lines cut ...]
  at android.database.sqlite.SQLiteConnectionPool.acquireConnection(SQLiteConnectionPool.java:348)
  at android.database.sqlite.SQLiteSession.acquireConnection(SQLiteSession.java:894)
  at android.database.sqlite.SQLiteSession.prepare(SQLiteSession.java:586)
  at android.database.sqlite.SQLiteProgram.<init>(SQLiteProgram.java:58)
  at android.database.sqlite.SQLiteQuery.<init>(SQLiteQuery.java:37)
  at android.database.sqlite.SQLiteDirectCursorDriver.query(SQLiteDirectCursorDriver.java:44)
  at android.database.sqlite.SQLiteDatabase.rawQueryWithFactory(SQLiteDatabase.java:1316)
  at android.database.sqlite.SQLiteDatabase.rawQuery(SQLiteDatabase.java:1255)
[... cut ...]
  at com.eveningoutpost.dexdrip.UtilityModels.MongoSendTask.doInBackground(MongoSendTask.java:41)
  at com.eveningoutpost.dexdrip.UtilityModels.MongoSendTask.doInBackground(MongoSendTask.java:21)
  at android.os.AsyncTask$2.call(AsyncTask.java:295)
  at java.util.concurrent.FutureTask.run(FutureTask.java:237)
  at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:234)
  at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
  at java.lang.Thread.run(Thread.java:818)

tzachi-dar, 2016-01-26:
>    android.database.sqlite.SQLiteProgram.(SQLiteProgram.java:58) at
>    android.database.sqlite.SQLiteQuery.(SQLiteQuery.java:37) at

Hi,

it seems that the logs that you have sent are not related to the problem of
send stuck.

Here is the way I have succeeded in getting the threads data on a not
rooted phone:

adb shell ps | grep -i xDrip
u0_a131   6267  183   897772 60848 ffffffff 00000000 S
com.eveningoutpost.dexdrip
adb shell run-as com.eveningoutpost.dexdrip kill -3 6267

adb pull /data/anr/traces.txt .

Please send the traces.txt file to me.

If this won't work, I will send you an apk that will be able to get stack
on a button press, and than we will get the data for sure.

Thanks
Tzachi

jmercay, 2016-01-26:
kill -3 says permission denied. I'll try again tonight.

jmercay, 2016-01-26:
ah I see the run-as now, that should help!

jmercay, 2016-01-27:
Looks like I need to rebuild the APK with android:debuggable="true"

$ adb shell run-as com.eveningoutpost.dexdrip kill -3 4608
run-as: Package 'com.eveningoutpost.dexdrip' is not debuggable

tzachi-dar, 2016-01-27:
Attached to this mail is an apk that you should install.
After the install, and when the send is stuck,

please go to properties ->Export CSV (suDiary format). once you choose that, the application will hang, and you will be asked to close it.
Now data should be in the traces.txt

run: 
adb pull /data/anr/traces.txt .

and send me the file.

Hope this time it will work.

Thanks

tzachi-dar, 2016-01-27:
[image]

Please rename the file app-release.apk and install it.

xJoe, 2016-01-28:
I will send you the file tomorrow as I need to use my private pc and I am not at home now.
Thanks for your help!

jmercay, 2016-02-11:
Can you re-attache that APK? I can't download it...

tzachi-dar, 2016-02-11:
Here is what you should do:

look at this thread in github: https://github.com/StephenBlackWasAlreadyTaken/xDrip/issues/131#issuecomment-172164491

on the post with app-release, choose "save link as..." and it will save it.

If this does not work, please send me your mail and I'll send it directly.

Cagier, 2017-04-18:
Just in case it might help someone else, I was having similar problems but it turned out that I had previously put some rubbish in the Dexcom Share settings and left the upload switched on.  The next time I changed the hardware source (from wifi + xbridge to just wifi) it then  started picking up the Share settings and timing out.  Switching back to wifi + xbridge did not fix the problem.  I needed to switch off the share upload and then restart the app.
Like I say, might help someone else troubleshooting and was probably not even the specific issue above...