StephenBlackWasAlreadyTaken/xDrip #132: Share receiver not showing up when scanning in 2.0.5 on Nexus 5 6.0.1

ryanthekiwi, 2016-01-16:
The Share receiver would not appear when scanning for BT devices in 2.0.5_1.  Tried with 2.0.3 and it worked great.

AdrianLxM, 2016-01-17:
@mgranberry, can you help with that?

AdrianLxM, 2016-01-21:
@ryanthekiwi, did you activate location services and allow it in the settings like described in the release notes for CM - might this be the issue?

ryanthekiwi, 2016-01-21:
I did not make any changes to the settings after installing the app.  My
location services are always on, but if an additional step is required to
specifically allow it for Xdrip in the settings I did not do that.  I just
downgraded to the older release and it worked straight away.

razzintown, 2016-02-02:
I can confirm that on my Nexus 6P running 6.0.1, this is fixed by going to Settings -> Apps -> xDrip -> Permissions and enabling Location.