StephenBlackWasAlreadyTaken/xDrip #135: Moto Display Notification on Moto X Pure 2015

aaroecker, 2016-02-18:
I recently switched phones to a Moto X Pure 2015.  The phone has a pretty cool feature called Moto Display this allows you to wave your hand over the phone to view your notifications without turning your phone on.  For some reason xDrip doesn't display as a notification for Moto Display if I select "Run Service in the Foreground" in the settings.   A link to the Moto Display app and more info can be found here:

https://play.google.com/store/apps/details?id=com.motorola.motodisplay&hl=en

xDrip will display in the notification area when the phone is on, but just not when using Moto Display.  When I use NightWatch the notifications are present in the regular notifications area with the phone on and also in Moto Display.  With that said, are notifications done differently with xDrip that would cause this, or could this be an issue specific to the Moto Display app?  Any suggestions on how to remedy this?

Thanks for your insight!

aaroecker, 2016-02-18:
Here's a video demo of what I'm talking about:

https://goo.gl/photos/QBe5Dtv8wNnC7Mfp8

Vaughanabe13, 2017-04-25:
FYI I have a phone with Moto Display as well and this is an issue with the phone itself, not xdrip. Moto Display will not show any permanent notifications, like weather, screen overlays, etc. xDrip is handled as a permanent notification when you check run in foreground. So unfortunately the two just aren't compatible at the moment.