StephenBlackWasAlreadyTaken/xDrip #136: Android M on LG G watch

msinkovich, 2016-02-20:
So without thinking I just swiped left to install the update yesterday.  Right off the bat everything seemed to work without issue but later in the afternoon the watch app started to fail.  A message would pop up that xDrip Prefs stopped working. 

At first to fix all I had to do was switch watch faces from xDrip and back again for it to start working again.  But when it happened again I was in the middle of something and couldnt do the "fix".  So after about 30 minutes when I tried it would only work again momentarily then back to error.  I dont have anything concrete as to a noticeable cause but am willing to send logs or what ever to help diagnose.  Just need to be pointed in the right direction.

thanks for all the time and work you guys put into this!

AdrianLxM, 2016-02-22:
Did you install the latest update to beta 5?

msinkovich, 2016-02-22:
I installed the update this morning.  I didnt see it when I logged in to
post this thread but Ill see if it solves the issue. From what I can tell
the crash happens when the watch and phone lose connection for too long.

AdrianLxM, 2016-02-22:
@msinkovich, that is a good sign as the bug fixed was one that happens if the connection was lost or no new values came in for longer than 16 minutes.