StephenBlackWasAlreadyTaken/xDrip #137: Bluetooth reconnect issue

porkupan, 2016-02-20:
Phone: Nexus 5; Android 6.0.1

Sorry if this has been discussed earlier.  The xDrip sometimes (rarely) loses connection with the receiver, and cannot reconnect for hours.   It appears that the phone is actually connected to the receiver, but the xDrip App reports "Not Connected".   I tried rebooting the receiver, turning BT off and on, hitting the "Refresh" button on System Status screen a bunch of times - doesn't seem to help.

Killing and restarting xDrip App appears to recover the connectivity.

The problem seems to occur when the phone and the receiver are located somewhere within BT range, but not really close.  For instance, when the phone is charging downstairs, and the receiver is in the upstairs bedroom.  So the connectivity may be somewhat marginal.   However, once in this state, the connection doesn't recover even in the close proximity.  I am not sure this problem can be readily reproduced, as it happens to us about once a week or so.

aaroecker, 2016-02-20:
I have experienced this too; I'm on the Moto X Pure.  I've been able to regain the connection by rebooting the phone.

michalkominek, 2016-02-23:
Hello to all.
After couple of weeks using xDrip I discovered very often missing packets from G4 sensor.
Up to 0,5h no receiving pockets. BT connection between phone and xDrip wixel works fine.
DEXCOM receiver is running without problems during this time.
Distance from G4 sensor to xDrip wixel is very short (0,3m.)
Do anybody have any experience (suggestions) in that matter?

aaroecker, 2016-02-23:
I believe that this is the issue, or is at least related to the one, that I recently closed; issue number 138.  I thought it was resolved, but it is not.  The bug seems to be triggered if one of two things happen.  One, if I leave my Dex and phone next two each other, but I walk too far out of range with the sensor.  When I come back into range the sensor reconnects with the Dex, but the new bg data is never uploaded to xDrip.  Clicking on "restart collector" from the system status screen resolves this, but usually the bug comes back even if I do not leave the sensor to Dex range.  Second, it's triggered if I  walk out of bluetooth range of my phone and Dex.  The bluetooth connection will reestablish when I come back into range, but again the bg data will not be collected.  the only sure fire way to resolve this is to reboot the phone and to keep all devices within range.

michalkominek, 2016-02-24:
Thx for answer.
I also suspect some „range” issues. That’s why I never go out of range (either sensor or BT).
xDrip and Phone always near the body.

[image]

msinkovich, 2016-02-24:
Where is the receiver in proximity to your phone?  I have noticied that I
need to keep other bluetooth things away from the receiver or else it will
drop signals....this has happend with my phone and PS4 controller being too
close....

porkupan, 2016-02-24:
I think we are confusing the fix with a workaround.  Yes, it's pretty clear that if the Dex and the Phone never leave each other's sight, then the issue doesn't occur.   For whatever reason, this suggestion doesn't always work, especially with the kids.  

The problem is that **when** the phone or the Dex is removed from close proximity, the App **sometimes** gets stuck, and never reconnects.  This doesn't happen every time or most of the time, which leads me to believe that maybe the disconnect has to occur at the moment when the data is actively being transferred from the Dex to the Phone.  Hopefully there is a timeout and retry logic of some sort.   Terminating and restarting the App fixes the issue.

Also, please don't confuse randomly missing sensor readings with this issue.  The G4 transmitter transmits a sensor reading once every 5 minutes no matter what.  If your receiver missed one, it will receive another one in 5 minutes.  The receiver sometimes gets confused about the signal quality or "noise", and stops displaying the BG for a while, but that's where we can take advantage of RAW.   However, none of this has anything to do with the BT reconnect issue I described.

michalkominek, 2016-02-26:
I I agree with both of you.
5-years old kid using mentioned xDrip. He has a pocket on the belly with equipment so it is not possible to go out of range.
Also I don’t think it is a problem with distance because it works very reliable few weeks ago. Transmitter battery =217 so it’s also OK.
I suspect some hardware issues with Wixel.

aaroecker, 2016-03-03:
Any update on this?

aaroecker, 2016-03-10:
I hate to keep asking, but is there anyone willing to help with this issue?  I've had several bluetooth disconnects this week over night for no apparent reason.  The phone and Dex sit right next to each other on the night stand.  This morning the disconnect occurred just before 5 am; I was still sleeping.  I was unable to regain the connection without forgetting my phone on my Dex, doing a clean install of xDrip and then establishing an entirely new connection.  This is becoming incredibly frustrating and concerning due to the disconnect occurring at night.  Thanks for any insight!

Choens, 2016-03-20:
I'll second this. My wife's setup has, in recent days, become very unstable. Dropping several times a day. If there's any useful log info I can submit, tell me what to grab and I'll gladly supply it to help debug this. I can also help with testing of proposed fixes, etc.

Choens, 2016-03-23:
Resetting everything (phone, receiver) seems to have helped, it only lost the connection twice in the last 24 hours. Still, less stable than it was.

aaroecker, 2016-03-23:
I've reset my connection so many times now I've lost count.  The bug eventually comes back for whatever reason.  I had a good run a few days ago; the Dex reconnected as it should and data was uploaded to xDrip, but it was short lived.  Under the Errors Log I get status bug 133 as the culprit.  The problem seems to occur with the Dex itself.  The phone under System Status states that the phone is connected to the Dex, but on the Dex the bluetooth icon is not visible.  When the Restart Collector button is pressed the bluetooth icon appears on the Dex and everything works as it should.  A simple solution to me seems to be that if no data is collected on the Dex for 6 minutes the collector should automatically restarted.  The root of the problem is why the Dex doesn't reconnect even though the phone thinks that it has.  Thoughts, @StephenBlackWasAlreadyTaken ?

abherrma, 2016-03-29:
I believe I'm having the same connection issues. connection will be good for 20-60 minutes and then lost. the only fix i found is to forgot BT connection on both devices and re-connect.

everything was working fine for weeks on my Droid Turbo 2. This started happening after my Droid Turbo 2 upgrade to android 6.0

@StephenBlackWasAlreadyTaken thoughts?

LJ1966, 2016-03-31:
Me too. Commenting so I can see updates.

After all my turning-it-off-turning-it-on attempts (several, over an hour, in the middle of the night) I left it turned off until the morning then tried again. It connected without problem. I spent a number of hours without any CGM data!

michalkominek, 2016-03-31:
No 100% clear solution. I have rebuild xDrip with new LiPo charger and it helps.
I suspect some sporadic voltage drops. 
Also I need to restart my Melrose S9 smartphone once per week. Now everything looks fine.

LJ1966, 2016-03-31:
I'm already using a LiPo charger in my xBridge. I have a bluetooth dropout every couple of days which is fixed by the off-on workaround. The other night was a major outage and nothing fixed it except time.

brekfest, 2016-04-07:
Just got a new phone running Android 6.0 (LG G4) and I'm losing connection every 10-20 minutes. Have to reset connection on phone and Dex to get it back. On old phone running 5.0 (also a G4), I very rarely got connection issues but could resolve simply by force stopping xDrip.

mountainsibray, 2016-04-16:
I'm seeing Bluetooth disconnect issues as well.  I just started using XDrip (awesome-looking tool), but am struggling with disconnects on the devices I've tried.  Moto G 2014 with Android 6.0 will only connect 1/10 times and then disconnects (gave app full permissions).  Moto G 2013 with Android 5.0 connects easily, but didn't leave it connected long enough to see if it was reliable.  Nexus 7 with Android 5.0 connects easily, but disconnects about every 30 minutes on average.

joshlrob, 2016-04-21:
I too started having Bluetooth issues after upgrading to Android 6.0.1. I tried forgetting the connection on the Dexcom share and xDrip and then repairing but the Dexcom never seems to think it is connected. Sometimes, immediately after pairing, the connection status does change to connected but no data is reported to xDrip. The Dexcom Bluetooth icon only shows during the pairing cycle and then shuts down. It is visible and broadcasting during that time. Bluetooth does work on many other devices with this phone so I believe the problem is between Dexcom and xDrip. Please let me know if I can help in any way.

gstoner, 2016-05-28:
I see the same issue with the current Beta release  2.0.5_2 with Andriod 6.01 on LG G Vista 2 with a Dexcom Share 2 Pediatric device.  ( we just had to replace our receiver, previous receiver worked fine with phone and Xdrip)   I enabled Settings -> Apps -> xDrip -> Permissions and enabling Location.  I have reconnected the device a number of times; double checked the ten character Dexcom Receiver serial number 

The new receiver is the same model number MT22608 same rev number LBL-0107785 rev05, just manufacutred 4 months latter  2015-08-06, vs 2015-04-17

Looking at the Error list, it is hitting DexShareCollectionService: Got the status 133 bug, bad news.....  already forgot the device twice and still running into the issue.

gstoner, 2016-05-28:
Dexcom Receiver Software is S10542 Rev 3.0.1.030,  I went back to try Version 2.01 of Xdrip 

Now I am seeing in the Error list 
ERRR GATT_READ_NOT_PERMITED2
ERRR GATT_READ_INVAILD_ATRIBUTE_LENGTH0
ERRR GATT_FAILURE 0
ERRR GATT_INVALID_OFFSET 2
ERRR GATT_INSUFFICENT_ENCRYPTION2
ERRR GATT_REQUEST_NOT_SUPPORTED 2
......

gstoner, 2016-05-28:
You also have error BgToSpeech teardownTTS()called but instance is null  with Beta release 2.0.5_2  that going on,    THe bluetooth discovery is working properly in eariler release, 2.05_2 is having hard time discovering the Dexcomm as bluetooth device.  I am going to try an older device Samsung GIII

gstoner, 2016-05-28:
Double check to see if the receiver can connect to Share2 app on iPHONE, everything works.    There is discovery and pairing issue with the current application.

gstoner, 2016-05-28:
I did find the phone can pair the Dexcom receiver when I use the Andriod Setting Menu--> Network-> Bluetooth Connections Available and select DEXCOMRX.   Resetting to factory setting to see if can get this to work again.

gstoner, 2016-05-29:
After factory reset,  Looking at the Error list, it is hitting DexShareCollectionService: 133 bug, bad news.....

kth100077, 2016-05-29:
Same issue here, and quite frustrating.  I've documented my experiences in issue #138, and here's the summary:
- xDrip current beta version on Droid Turbo.  Dexcom G4 with Share.
- Collector stops every few hours and no data is gathered until I manually re-enter the System Status menu and re-start it.
- Proximity is not the issue.  My G4 is within two feet of the Android device during the entire time.
- Shutting off wifi and/or other bluetooth activities is not a useful workaround, so I haven't bothered trying it.
- I've tried several proposed fixes (see [Issue #138 ](https://github.com/StephenBlackWasAlreadyTaken/xDrip/issues/138#issuecomment-221340986)) without success
- The Dexcomdevice continues to perform beautifully with or without xDrip pairing.

I was insanely happy to have Dexcom data on my phone, where I would see it far more often than on my Dexcom device.  But after this bug appeared, it's no longer fun.  

I have seen 133 bug listed in the error log, but i have no idea what it means ... let alone what do to about it.

hardaker, 2016-05-31:
Adding in my own data here, some of which I think is semi-unique:

1) I completely wiped xdrip data from my phone hoping a complete restart would help (saving an archive, though crying over the upcoming loss in 90 day A1C predictions).  Unfortunately, the solution didn't work and it lost connection not too many hours alter after the complete reset.

2) Having given up on manual re-connections (because a few hours on and many off seemed pointless), I left the app to doing its own thing and ignoring it.  One thing I'll note in that the connection occasionally comes back for a few hours by itself.

Those are the only new things I have to offer, as everything else mirrors what people have said above: It was fantastic when it worked and I was in love.  It doesn't seem distance dependent.  etc.

Cagier, 2016-06-09:
I too seem to have a unique issue but it is bluetooth related so I will add it here for the moment rather than raising a new issue.

Phone: Moto E (2nd Gen) - has worked for over a year
Android version: 6.0
Android Security Patch Level: February 1, 2016
xDrip App version: beta/2.0.5_2-7a93a06-experimental-xDrip2-beta-205-2-2016.02.20

I have been having increasing problems with missed readings and have found that restarting the collector seemed to fix these temporarily.  Recently when I restarted the collector it did not connect again.  I tried a BT scan but not there either so I tried Forget Device and since then I have not been able to reconnect to the xbridge.

Another strange thing happened at the same time.  My other phone, which is running an xdrip wifi immediately kept grabbing the xdrip BT connection.  Even when in wifi mode and not linked by bluetooth, I can see the BT light on the xdrip come on and off as I toggle bluetooth on that phone.  It is not connected via the android bluetooth devices and is only remembered as an xdrip device.  There do not appear to be any other apps trying to connect to it.  However, even when this phone is switched off, my original phone still refuses to connect as it still cannot see it from within the xdrip app.

However, I can see the xdrip under the android bluetooth connections but it will not connect as it prompts for a PIN.  Sometimes I get less error messages if I use 000000 but it still does not connect.  However, if I go into the xdrip app and do a BT scan it still will not find it and so will not connect to it.

The xdrip is actually working fine because if I switch my other phone (a HTC One) from WiFi wixel to xbridge wixel then it appears straight away there and can pick up BG readings.  This phone can also connect in the android bluetooth screen but unlike the Moto it does not prompt for a PIN.  I am using a slightly different version of the xdrip app on the HTC (1af3b7e-experimental-xDrip2-beta-205-2-2016.01.12) but I don't think that is the problem as the other version has worked fine on the Moto for ages.

So going back to the original Moto phone, when I switch the xdrip into Wifi wixel mode it works fine so it seems to be purely a bluetooth issue.  However, bluetooth still works on the phone for other things (such as file transfers).  I also tried just powering up another BLE module with a couple of AA batteries and this cannot be found within the xDrip app either so it does not appear to be a specific problem associated with it remembering/forgetting my xDrip (although both BLEs have the same name).

I've looked in the error log but nothing to see there after an unsuccessful bluetooth scan.

I have also tried loading other xdrip versions but I just get an error saying the app failed to install.  It is really frustrating.  I am loathe to wipe the phone and rebuild again as from comments above this never seems to fix anything.  I also do not want to lose all my history by uninstalling as it would screw up the historical stats etc.

If anyone has any other suggestions they would be really welcome!  If all else fails I will go for a wipe and reinstall but I think I will be just more depressed when that doesn't fix anything...

Cagier, 2016-06-12:
OK, as no suggestions to above, I have done a factory reset and installed a different version of the xDrip.  Still cannot see it in a bluetooth scan within the xDrip app, although my other phone can.  I can't help feeling that this is an android-related issue but as I said I can see it in a bluetooth scan under the android bluetooth settings so it is only the xDrip app that cannot see it.  The xDrip app still works with wifi devices so using for that at the moment but will have to try buying a new phone to overcome this software issue unless anyone has a better idea?

nztianmi, 2016-06-12:
If your phone is running Android 6 you need give location permission to xdrip. Other can't see it drom bluetooth.  Worked for me

Cagier, 2016-06-12:
Thank you so much!  I had seen reference to that before and had looked under location but didn't see anything obvious there.  I did search the repository for details on how to do this and googled and downloaded various "permission managers" but it did not appear.  Eventually tried the latest xdrip experimental version and unlike a previous version this did appear under Apps/Settings(cogwheel)/App Permissions/Location and I could toggle it on there.  Then everything works again!
Brilliant!  Thanks again...

eurenda, 2016-08-23:
My Samsung Note 4 with Android 6.0.1 would initially stay connected to dexcom G4  with xdrip beta/2.0.5_2 and intermittently disconnect but I could reconnect by restarting collector under system status in xdrip.   After having issues reconnecting I decided to forget the device from xdrip, dexcom, and unpair from phone.... and here the tail of woe begins. 

I can no longer maintain a bluetooth connection even though I forget the device on xdrip, phone,  and dexcom.  I am able to see the device,  have gps location enabled,  and add the dexcom as a trusted device.   I can reliably,  forget and rediscover the device and maintain connection for about 20-30 minutes before receiving the "DexcomShareCollectionService Got the status 133 bug, bad news" error in the logs from which no other data is received - only way to come back is to forget/unpair devices on dexcom, xdrip, and phone bluetooth.  Sounds like a known issue,  but wanted to post details in case it helps troubleshoot.

c-nottingham, 2017-01-01:
@eurenda @hardaker  did you ever get your Note 4 to work again?

I'm having similar issues with both our Note 4's. It isn't a range issues, the devices are always within 5 feet of each other. My son's had been working perfectly up until about 48 hours ago when it started dropping bluetooth about every 20-30 minutes. I checked his logs and we are getting the code 133, bad news" errors. We've tried uninstalling and switching xDrip builds, we know about the location issue and make sure it has all perms, cleared the Bluetooth data and cache, disconnected everything (G4 with Share) and reconnected more times than I can count.

When it wouldn't work on my son's phone, we tried it with an old Apple phone, which seems to work fine. The we tried my Note 4 and it drops the signal in the same way. At this point, the kid is considering a factory reset (since it had been working for months) but I kinda don't want to deal with the fallout if that doesn't help (and it doesn't look like it helped others).

Has anyone been able to solve their issue?

grincher, 2017-01-02:
when my family member started getting disconnects and 133 bad news errors
(Moto X 2nd gen & g4share) I tried all the same (forgetting, clearing,
installing different xDrip versions).  what seems to have worked is to
uninstall xDrip and install xDrip+.

hardaker, 2017-01-06:
@c-nottingham : as mentioned above, I ended up with a new phone entirely (never had a note4; I switched from an S5 to an S7).  Anyway, my current phone works for long periods of time and then suddenly stops.  But unlike my S5, when I reboot the S7 the problem goes away and it reconnects again.

jreneefriedberg, 2017-04-27:
this sameish issue happens with me with my dexcom transmitter linked to an iphone.. i'm not sure what it is.. but the sensor cuts out for more than 3hrs (general longest its supposed to) can call dexcom and get a replacement sensor.

i've tried the following with apple:
close all apps, reopen dexcom, turn off/on bluetooth, reboot phone..
sometimes if it doesnt connect the first time it's inserted at all.. ill make sure there is only one bluetooth connection.. the extras (due to deleting / redownloading the apl) creates new bluetooth connections that interfere with the connected one..

just saying apple to dexcom has the same stuff going on.. maybe some simular things will assist in fixing the problem sometimes

bashfoo, 2017-05-30:
My son is running most recent xDrip app on Samsung 6 active with G4 Dexcom Share. We are experiencing the same problems with phone, Dex, and transmitter all in same room, within feet of each other. While the Android OS will connect with the DEXCOMRX device time after time, the xDrip app will ignore that established connection. Almost like it's fighting for control of it? 
The oddest thing is that the phone BT connection seems solid. It sees the device 100% of the time. It recognizes it and will establish a connection to it. However the System Status in xDrip reports not connected. Refresh, Forget Device, reload app, restart phone.. nothing really works. Occasionally, I can get it reconnected after many attempts at forget device, scan for device, and then refresh connection on System Status screen. 
I would think that using Wifi would be so much more reliable of a connection, has anyone stumbled over one of those project devices that can replace the Dexcom Share and possibly provide WiFi connection to the Phone and Internet? I'd imagine that would be so much easier to troubleshoot and validate connection. It almost seems that the Dexcom Share device doesn't have enough juice to retain the BT connection with the phone. And.. Apparently doesen't have a really robust 'reconnect" protocol for when the Dexcom share and Android phone part ways, even temporarily.

kews12, 2017-06-01:
Im having the same issue using xDrip beta 6 2.0.2_1 on Moto Droid turbo 2 Android Nougat (7) with G4 Share. Can pair and start getting readings but bluetooth disconnects after 7-10 mins. The Dexcom receiver shows no bluetooth icon but phone shows connected & xdrip says no data after that initial time period, gives the error 133. Ive turned off all power saving & can get to reconnect by resetting everything but only for less than 10 mins. The receiver & phone right next to each other & havenet moved them. Im finding this issue goes back over a year and on the previous android version, but no resolution for many.

jweismann, 2017-07-02:
Hi, after reading this thread  I realized that I could have posted my issue here as well. Maybe you will find my issue relevant https://github.com/jstevensog/wixel-sdk/issues/26 for this thread too.

Cheers, Jacob

Mogsternut1, 2020-06-16:
Lots of posts all problems as with me. Just 3 days in, a newby using xdrip+. Shame it doesnt work any more. Not sure where to ho from here besides wait.