StephenBlackWasAlreadyTaken/xDrip #138: Bluetooth disconnect

aaroecker, 2016-02-21:
My phone has been disconnecting from my Dex a lot lately.  I have a 2015 Moto X Pure running Android 6.0, I have battery optimization off for xDrip and location services enabled.  The disconnect occurs mostly at night which leads me to believe that it has something to do with doze, but it did occur this morning as well.  At night my phone and Dex are right next to each other, so there shouldn't be any connection issue due to distance.  Last night it lost the connection around 1am and never reestablished a new connection.  The real trouble is that I can only get a new connection again if I forget the device on my phone and Dex and reestablish an entirely new connection.  If I try to just scan for Bluetooth devices the Dex will appear, but will never connect.

Any advice on how to troubleshoot this?  It seems very similar to the issue #137, but a little different.  Thank you for any insight!

aaroecker, 2016-02-21:
Any insight into this my Dex has disconnect over a dozen times today and the only fix seems to be by opting to forget the devices on the phone and Dex and reestablish a new connection.  I'm starting to wonder if this is a hardware issue?  Thanks for your help!

aaroecker, 2016-02-21:
I may be wrong about the Bluetooth connection being the issue.  Xdrip was showing signal loss as the alert message, but it showed being connected the my Dex within the app.  I clicked on restart collector and it regained the signal.  Could the app be freezing up while collecting BG data?  I apologize for all the updates, but this has been frustrating me all day.

LadyViktoria, 2016-02-21:
do you use the newest xbridge firmware on the wixel?

aaroecker, 2016-02-22:
I have the G4 w/share and am connecting the Dex directly to my 2015 Moto X Pure.  The issue seems to be really that the app stops collecting my BG data.  Additionally, I've been troubleshooting it and it seems device specific.  I tired setting up xDrip on my wife's Moto X and the same issue occurred; data collection stop, but the Bluetooth connection stayed connected.  I then set xDrip up with my 2013 Nexus 7 tablet and it never stopped collecting my BG data.  The odd thing is this only started to occur recently, so I'm wondering if a factory reset is in order on my Dex and phone.  Is there a way for me to a log or bug report?  I don't know how to do it so someone would have to talk me through it.

aaroecker, 2016-02-22:
Yes, I'm using the latest xDrip beta.

aaroecker, 2016-02-22:
After a factory reset and trying different versions of the xDrip I was able to get things working again.  I really have no idea as to what was causing the issue, but after a lot of frustration things seem to be back to normal.  I'm currently running the latest beta without issue.  Here's hoping the issue doesn't show up again.

LadyViktoria, 2016-02-22:
ok thank you
then we can close this issue?

closed 2016-02-22 by aaroecker

kth100077, 2016-05-24:
This issue is ongoing.  I am on Android 5.1 with a Dexcom G4/Share.  So far I am only using xDrip to display the CGM data on my phone, but its unreliability makes it unusable.

xDrip was extremely stable for my first month and a delight to use, but now the collector crashes constantly -- within minutes or at most hours of restarting.  I have reinstalled xDrip, had both devices forget each other, cleared the system cache, and then reconnected everything.  The problem continues.

I've had advice to turn off wifi, but disabling essential phone features is not a satisfactory answer.

Diagnostic info:
The only clue I have is that the Droid Turbo got a minor system update in the last two weeks, possibly coincident with this problem.  However, the fact that so many others have had similar problems leads me to think that the issue is deeper.

FWIW, the bluetooth icon on the Dexcom disappears when the collector is off. Share is still on from the point of view of the Dex settings.  When I restart the collector on the phone, the Dex Bluetooth item reappears.

Thanks for any help!

StephenBlackWasAlreadyTaken, 2016-05-24:
Have you restarted the sensor session in xDrip? If not your database could
be under too much stress

kth100077, 2016-05-24:
Interesting.  Does "Stop Sensor" have a different meaning in the xDrip menus (stop/restart a sensor **session**) vs. in the Dex settings (stop/restart a **sensor**)?  If so I will try it.  But that's ambiguous language that might be cleaned up in the UI.

Thanks for the fast hint!

StephenBlackWasAlreadyTaken, 2016-05-24:
> When I hit "stop sensor" on the xDrip UI, it gives me a warning suggesting
> that it's actually going to stop the sensor in the Dexcom sense. Hmmm. And
> if I were to do it at the time of sensor replacement, what sequence should
> I use for stop Dex, stop xDrip, start Dex, start xDrip? Is this documented
> somewhere I missed?

You should be stopping it in xDrip when you are actually removing the
transmitter from the sensor. The order does not matter all that much but
that's the idea.

If you are using xDrip with the wixel or the  g5 then it will impact your
calibrations when you stop and start it but if using it with the share
there should be no noticeable changes other than it not growing out if
control

StephenBlackWasAlreadyTaken, 2016-05-24:
(so if you are using it with share you can feel free to stop and start it
now, but try and get in the habit of doing it just when Changing the sensor
out)

kth100077, 2016-05-24:
I'll try that now, Stephen.  Yes, I'm using it with Share.  

How much data starts to choke the database?  A week? Several weeks?

FYI: I am stretching my sensors to last two or three weeks, as long as the adhesive works OK and results are not weird.  I just use the Dexcom settings to stop and then restart the sensor without actually changing it. 

So if I didn't have Share, when you say I should stop it in xDrip "when you are actually removing the sensor."  I suppose more precisely you mean "when you stop the sensor using the Dexcom" even if I'm not going to remove the sensor, just restart a few seconds later.  Am I absorbing this info correctly?

Many thanks for the quick and knowledgeable replies!

StephenBlackWasAlreadyTaken, 2016-05-24:
You will want to do it when you actually pull the sensor out of yourself.
the amount of data it can handle before it starts to choke is dependent on
the phone but most phones should be able to handle a month or so without
much trouble.
The tiiming/ordering of the starting and stopping things isnt all that
important for a share user, but the rule of thumb is to stop before you are
going to take it out and start just after you put the new one in

kth100077, 2016-05-25:
> You will want to do it when you actually pull the sensor out of yourself.
> the amount of data it can handle before it starts to choke is dependent on
> the phone but most phones should be able to [...]

Well this was better than before, but did not solve the problem.  Stopped the sensor just before noon ET, and it ran uninterrupted until 10:15 PM, which is nice.  But now I've re-started the sensor twice and it's still not collecting data.

So I don't think I have the full answer yet.  :-(  I sure love having the data on my phone where I see it several times an hour.  Wish I could keep it going.

kth100077, 2016-05-29:
This bug continues.  in the last week I've completely uninstalled xDrip from my Android device, cleared the Android system cache from the boot menu, forgotten all Bluetooth connections from the Dexcom, and started fresh.  Within 2 hours xDrip stops collecting data.

Such a shame.  I had high hopes for this software, but it either has lethal bugs in it or the documentation of how to connect and troubleshoot is insufficient. 

Sadly, I've deleted xDrip.