StephenBlackWasAlreadyTaken/xDrip #139: Bluetooth Module Support

apabari, 2016-03-18:
Hi,

I'm in Canada and have been struggling to get a hold of an HM-10 BLE module.  Have been waiting two months for the ones I ordered from e-bay, and still no arrival.

Are there any other BLE modules that will work with the xDrip setup?  

Canada Robotix has quite a few in stock... 
http://www.canadarobotix.com/index.php?route=product/search&search=bluetooth

I picked up the Adafruit Bluefruit EZ-Link-Bluetooth to see if it would work, and it did not.  

I'm ITCHING to get this built and want to pick up a BLE module in person today... just not sure which one to get, or if any will work!

Any help is appreciated.

Thanks all,

Tatt2edone, 2016-03-29:
I have struggled as well with finding the module, I will be ordering some from an overseas company I use quite often so I know I can trust them. 

I am ordering both HM-10 and HM-11 as they both have been reported to work.

I am new to pcb and boards and modules so I have no idea what other parts may be a suitable alternatives.

I can get you the product links if you wish. Maybe if we're close errr can compare notes and finished xDrips.

StephenBlackWasAlreadyTaken, 2016-03-29:
Just about any hm10 and hm11 will work, the ones without pins are difficult
to solder (for me at least!) but are generally easier to find!

Unfortunately I have not added any other modules into the mix (I started on
the redbearlab but never finished)

jamesthurlow, 2016-04-24:
Hi i'm creating an NFC reader from a LIbre to X-drip.  Just noticed this after I couldn't get the Adafruit Feather to connect to X-drip.  Does X-drip just search and connect to HM10/HM11's

Was also looking at the Redbearlab - nice small unit, shame it lacks the integrated charging like the feather.

StephenBlackWasAlreadyTaken, 2016-04-24:
I have not seen the feather, I'll check it out and see what might need to
be done!

StephenBlackWasAlreadyTaken, 2016-04-24:
So it looks awesome, really awesome, thanks for pointing me towards it!
I'll pick one or two up!

jamesthurlow, 2016-04-24:
Think best just ignore me :-) the adafruit app was causing the problem - put x-drip on a tablet with no adafruit app and it sees it fine..  not got it working yet as feather uses the SPI bus to manage the bluetooth which is conflicting with the NFC reader..   That Redbear Nano is cheeky little unit though http://redbearlab.com/blenano/

StephenBlackWasAlreadyTaken, 2016-04-24:
I do have a few nanos but I never got around to using them for anything
yet!

Tatt2edone, 2016-05-01:
Hey everyone, so I have successfully built the bridge with a bare hm-10, those little half cups suck to solder. On my second one as the first module list a cup when the txd wire got a tug. The red bear nano looks nice, but my question is... Does the bt module only have to be bt to UART? 4.0ble I know, but I am curious. I want to build a bunch to give out at cost to people in the T1 community here in Ontario but looking for cost effective parts and I am just not overly happy with the hm-10 half cups.

What's this feather thing? Does it work?

jamesthurlow, 2016-05-02:
Tatt - sounds like your intentions are hounerable but would highly recommend you exercise caution.  Building these kits for others you are by implication creating a degree of liability for yourself.  I'm sure if you posted on the CGM in the cloud FB group you would get some direction / advice.

The advantage with the feather is that it also has a charging circuit (the BLE UART does not) in theory it has all of the components for an Xdrip without the need for a soldering iron.  Thing is the code needs to be adjusted to suit this device as it manages the bluetooth differently as the bluetooth is not seperate.  I'm sure Stephen or someone else will get their.  I don't use a Dex so can't help sorry.

Tatt2edone, 2016-05-03:
Thanks for the reply James, I get the concern with building for other people, from what I have seen and read, it is quite common to have "xDrip" Parties to build a bunch for others. I am a member in a couple FB groups, until last night, for some odd reason, no one shared any source for parts. No one seems to be trying out other boards or methods, so I though this would be the best place to look.

I am not knowledgeable in program languages, so I am hesitant to try the RedBear Nano or Feather if it has to be programmed, but I'd really like to find a good solid platform, easier to obtain parts list, even if just for my wife's xDrip.

thecubical, 2016-05-03:
I found the following 

https://github.com/mzst123/Xdrip-Lipo-Board
https://github.com/IdleHandsProject/xDrip_PCB

Still require some soldering though.

Id like to see an xdrip with wifi support that can remain plugged in, I think there is a feather version that has wifi but I don't know enough about them to know if that one can also read the wifi signal.

What exactly does a board have to support to be able to read the dexcom signal?

Tatt2edone, 2016-05-03:
I saw an HM-10 board to mount the bare HM-10 on at OshPark, that one I can figure out, but the first link in your post, I have no idea what I would do with it, or the parts list..haha

This is all new to me, until this weekend, the only soldering I have done is plumbing pipes...I have a few electrical controls friends who are trying to help out finding ways to make it more compact and/or sturdy...

Thanks for the links, I will look into them further for sure!

thecubical, 2016-05-03:
Theres also this guy http://xdripkit.co.uk/ I did enquire about it but for the price it was easier for me to just solder it myself, his kit still requires a little soldering but much less and its all nicely packed together.

Tatt2edone, 2016-05-03:
I don't mind the soldering, I love to tinker, I am fairly techie, just more of a phone/tablet home media nerd. I used the xdrip uk site to get the program file generated for me, no editing needed, just download, and load to wixel.
My main issue is finding an HM-10/11 alternative that is locally available, at least available in Canada anyway, I can get almost everything here already, but so many unknowns when it comes to the Bluetooth module...I have yet to find out the exact reason why it has to be HM-10/11 and the cc2541 chip. 
I looked at the Feather its all in one, it'd be sweet, I just have NO idea how to begin programming it to work like the wixel and still communicate like hm-10.

thecubical, 2016-05-03:
Pretty sure this should work

http://www.ebay.ca/itm/HM-10-BLE-Bluetooth-4-0-CC2540-CC2541-Serial-Wireless-Module-Arduino-Android-IOS-/141777677617?hash=item21029bad31:g:EkkAAOSwsB9V-oc4#shpCntId

It looks like the exact one I have,

thecubical, 2016-05-03:
Can anyone confirm if the feather would actually work, is it able to communicate via BLE and also listen for the data the dexcom is sending?

nicolasdexcom, 2016-10-16:
Hi....Im i ontario and have an xdrip. The bluetooth signal says HM-soft. I have been told that i have to re-program the bluetooth. 
Can someone help me with that? The bluetooth should read xbridge but it doesnt. It is still picking up sensor but the readings are way off. 
Any help would be awesome. Thanks

amlynek, 2017-06-26:
http://www.robotshop.com/ca/en/xdrip-bridge-module.html#What
For those of you looking for parts or the entire xBridge kit. I have purchased two of these from this site. No issues at all. FYI, I am not affiliated with the site in any way.