StephenBlackWasAlreadyTaken/xDrip #140: Treatment interface possible?

porkupan, 2016-03-21:
I was wondering if any of the developers would consider implementing a basic treatment interface?  Since most setups don't have access to the pump data, maybe XDrip could implement manual bolus information entry similar to the Nightscout Web interface?  My son does a lot of correction boluses during the day at school, maybe he would bother recording those boluses via XDrip if such an interface was quick and simple to use.  Currently there is no way of knowing why the BG is going in the direction it's going.  

I am trying to figure out the reason for persistent after-lunch high BG at school.  Seems that cannot be beaten down by multiple correction boluses.

Also, would be nice to provide RAW display option.  To see the RAW data the XDrip user needs to load the Web Interface - this seems pretty redundant.

whoiswes, 2016-03-21:
xDrip+ (a fork of xDrip) does treatments via voice detection:  https://github.com/jamorham/xDrip-plus

porkupan, 2016-03-21:
Wow, thanks, this is great to know!

porkupan, 2016-03-21:
Is XDrip+ binary available to download, or do I need to build it?

whoiswes, 2016-03-22:
Download available here:  https://jamorham.github.io/