StephenBlackWasAlreadyTaken/xDrip #144: Expirimental APK for use with G5 Transmitter?

briancmoses, 2016-03-30:
Hey all,  this is [/u/neckhole](https://www.reddit.com/user/neckhole) Just following some instructions from [/r/Diabetes](reddit.com/r/diabetes).  Hoping that you guys wouldn't mind emailing me the latest experimental APK and I'll do my best to help put it through it's paces.

Thanks!

nicking44, 2016-03-30:
Have you gotten the apk?   I have yet to receive an email for it.

briancmoses, 2016-03-30:
Nick,

No, I haven't received the APK yet.

deculpep, 2016-03-31:
I'll also give it a go on Moto G 2015 3rd gen running android 6.0

nick5768, 2016-04-12:
I would also be willing to test this on a Samsung Galaxy S7.

taylorfowler, 2016-05-02:
@briancmoses / all  Any luck here guys? could build the apk too, if necessary. Is there a separate branch for the G5 apk?

briancmoses, 2016-05-02:
@taylorfowler No luck so far,  I'm assuming it's pretty low-priority and/or may have been inadvertently forgotten about.

deculpep, 2016-05-02:
you can build from the source at https://github.com/StephenBlackWasAlreadyTaken/xDrip-Experimental/releases/tag/alphaG5%2F2.0.6_1

taylorfowler, 2016-05-02:
@deculpep  Rock. Thanks!

msinkovich, 2016-05-24:
Just got the dreaded low battery notice today...gonna have to break out my G5 soon....any new updates???