StephenBlackWasAlreadyTaken/xDrip #145: [G5 Experimental] Transmitter not connecting to phone

nicking44, 2016-04-02:


nicking44, 2016-04-02:
Phone - HTC one M9
Put Transmitter ID/ SN number in correctly
after the 120 minute (2 hour) mark says "Please wait, need 2 readings from transmitter first"  and doesn't change after it hits that
I'm not sure if it's connecting to my phone or not, it's just not displaying any info on it

System Status
Collection - G5
remembered device- not set
connection status- Not connected
Transmitter battery- not available

gwoody1984, 2017-07-31:
I'm having the same issue, did you get that resolved?

apeterson22, 2017-07-31:
I have not gotten mine to connect at all.. I've even factory reset my phone
and installed the latest nightly. Logs are not even showing the attempt to
scan and find.

nicking44, 2017-08-01:
@gwoody1984 @apeterson22 I did not since my transmitter ending up dying shortly after this post, and my supplier has been a pain about getting a new one (should have one in soon though). Can you make sure that location is also enabled since I was having a similar issue with the Pebble and that fixed it. Also what android version are you on?

apeterson22, 2017-08-01:
@gwoody1984 <https://github.com/gwoody1984> @apeterson22
<https://github.com/apeterson22> @nicking44 <https://github.com/nicking44> The
SM-J320A is on 6.0.1 (I believe). I'll double check that when I home later
today. Location is on, I've cycled through all the setting,
uninstalled/reinstalled app. Factory reset phone and reinstalled app. No
luck. It's not even trying to scan for the device (at least what the logs
are saying, the save logs feature is grayed out so I can save them other
then what get sent). I do notice this error coming up as I'm cycling
through the setting and getting the initateScan entry of ( rollCall Error
loading index: java.lang.NullPointerException: Attempt to get length of
null array). I have a different (cheaper due to Frys having it on sale)
phone that I'm also going to try it on and see if I can get it to work. If
all else fails I will run it back on my phone (Galaxy S6) since I know it
works there (I don't keep it on that phone because this is for my 5 yo).
I'll get the rest of the specs later today on the phone. FYI the "cheaper"
phone I will try it on is a Blu Tank 4.0 Extreme.

apeterson22, 2017-08-01:
Model Number: SAMSUNG-SM-J320A
Android Version: 6.0.1
Android Security Patch: October 1, 2016
Baseband Version: J320AUES2AP2
Kernel Version:
	3.10.9-8241677
	DPI@SWDD6915#1
	Tue Oct 18 16:40:12 KST 2016
Build Number: MMB29K.J320AUES2APJ2
Security Enhancement for Android:
	Enforcing
	SEPF_SECMOBILE_6.0.1_0029
	Tue Oct 18 17:01:43 2016
Security Software version: ASKS v1.1_160901
KNOX Version: 2.6
	Shared Devices 2.6.0
	TIMA 3.2.005
	CEP 2.0.1

XDRIP+ settings
Version: dc550b1-2017.07.03
	Code:170703202
	Downgradable to: 1603091400
Collection Method: DexcomG5
Bluetooth Device: none Set
Connection Status:	Not connected
Phone Service State: Started: 14:47
Transmitter ID: <this is correct>

nicking44, 2017-08-02:
https://github.com/StephenBlackWasAlreadyTaken/xDrip/wiki/xDrip-with-Dexcom-Share 
this should cover the basic of setting it up with any device just change share to G5 and you should be set.

okay going to go over some basic setting since sometimes those are the ones people miss.

I'm on a custom 7.1.2   so my setting might differ from yours a bit. 

1. Configured Bluetooth to be on and location service enabled, (haven't been on 6 in a while so I forget when this was added)

2. configured setting correctly (device ID)

3, scan for device

4. toy with some other settings if still not connecting

if you can, can you post screenshots of the settings and device setting (in xdrip, (don't quite remember what the selections were called, as I don't have it installed currently) all would be good). as well as the errors (1-2  full screen should be sufficient for me to see what's happening). I might spot something that was missed or you might not of thought of.

gwoody1984, 2017-08-02:
Ahhh, I am working off of latest code in master. Didn't realize that didn't support the G5. I'll have to look at the latest beta and see if that works. Do you know if that code is also in this repo?

My device:
Nexus 5
Android: 6.0.1

nicking44, 2017-08-02:
The Beta version supports the G5 and that's what I usually use. although there may be some hicups with it at times when I had it working it worked (i'd say) about 90% of the time with some potential missed reading here and there trough out the day

here is the link to the version that supports the G5 

https://github.com/StephenBlackWasAlreadyTaken/xDrip/wiki/xDrip-Beta

taylorfowler, 2017-08-02:
You guys may want to check out: https://github.com/NightscoutFoundation/xDrip 

This code base is forked from here, but is actively being developed and may resolve some of your issues.

apeterson22, 2017-08-02:
> NightscoutFoundation/xDrip

I was using https://github.com/NightscoutFoundation/xDrip when it stopped
working and would not start back up. Then I upgraded to the Latest Alpha,
same issue, then I uninstalled and installed https://github.com/
StephenBlackWasAlreadyTaken/xDrip/wiki/xDrip-Beta still the phone wasn't
working.

taylorfowler, 2017-08-02:
Try popping in here: https://gitter.im/jamorham/xDrip-plus

Also, it looks like you havent set the transmitter ID?

nicking44, 2017-08-02:
@taylorfowler  if the " haven't set the transmitter ID" was directed at me I made sure it was set since when I made this issue it was on a new application and it stopped working, the transmitter remained the same from the last.

@apeterson22 did you confirm that all setting were inputted correctly? transmitter ID? might want to select constantly scan or force foreground (I believe that's what it's called) to make sure that the phone isn't forcing it to close the Bluetooth aspect of the application.

apeterson22, 2017-08-02:
Yes. I've cycle those on let attempt for a few hours, cycled through all
the setting.

nicking44, 2017-08-03:
just reinstalled not long ago personally so I can give detailed instructions to help out.

Can you post error page? located at "Xdrip>setting>Other settings>View Recent Error/Warnings", its almost at the very bottom of the list

so to confirm
You've tried both 
https://github.com/NightscoutFoundation/xDrip Which was working until it stopped, was that after changing transmitter, or just suddenly stopping? you might need to change transmitter, if that's been active for ~3 months

https://github.com/StephenBlackWasAlreadyTaken/xDrip/wiki/xDrip-Beta (what I'm on)

you've also tried
 multiple other settings like forcing UI thread, Run in foreground, Scan constantly, etc?

apeterson22, 2017-08-03:
> Here is QR code
> So I bailed on the j320a and I now have the Blu tank extreme 4 connected
> and working. For now at least. Had to play with more phone setting then
> xdrip+ settings for that..
> Yes the transmitter is only a month old and it was working on Saturday
> then Sunday evening just stopped picking up on the j320a. That was on the
> nightscout release and was never able to get it to pick back up..

Won't let me upload images via mail, will log into my computer later and
attach or send a drive link..

apeterson22, 2017-08-03:
So I bailed on the j320a and I now have the Blu tank extreme 4 connected and working. For now at least. Had to play with more phone setting then xdrip+ settings for that.. 

Yes the transmitter is only a month old and it was working on Saturday then Sunday evening just stopped picking up on the j320a. That was on the nightscout release and was never able to get it to pick back up.. 

 Will start in checking the experimental ones tomorrow once it's been running successfully for 24 hours. I will be returning the J320A phone since it's not needed and the Blu phone is more 5 yo friendly.. I have also uploaded the logs via the app several times times to get all the errors.

jamorham, 2017-08-03:
The xdrip-beta version from this repository will definitely not work on any g5 transmitter manufactured after nov 2016. I'm not aware of any circumstances where that version will work in preference to the nightscout xdrip+ version which is here: https://github.com/NightscoutFoundation/xDrip

nicking44, 2017-08-03:
@jamorham
If that's true I'll install it now and if it doesn't work I'll move back over to this one and test it on this. Thanks for the heads up.

apeterson22, 2017-08-03:
I installed the nightly since I couldn't get the Beta or the nightscout
versions to work.. The nightly is definitely working on the Blu Tank Xtreme
4 phone. From the Status page:
dc55ob1-2017.07.03
Code: 1707032102
Downgradable to: 1603091400
Connection Method: DexcomG5
Connection Status: DexcomM4 Authed
                                FW: 1.0.4.10

apeterson22, 2017-08-03:
[image]

[image]
[image]
[image]

apeterson22, 2017-08-03:
This is the initial error that I get when installing and configuring. 

[image]

nicking44, 2017-08-03:
@apeterson22 I havn't seen that error before maybe @StephenBlackWasAlreadyTaken can help you in what that means, but to me it looks like something isn't supported or executing as it should be. Was that the same issue on both of the other versions you tried?  might be what @jamorham said, in it not working with dexcom g5 made after Nov.2016. but glad you got something working for the time being. 

I'll be able to test mine today when they arrive so if I have the same issue I'll try that one and see if it works for me too.

apeterson22, 2017-08-03:
Here is an update for the nightly. My son walked away from the phone, and it lost connection for about 28 minutes. It would not reconnect even though the logs showed as bonded and scanning. Restarting the collector did not allow it to reconnect. Missing about an hour of data, I restarted the phone, and it started picking up and working again.. I noticed the RollCall error occured once during the time he was out of range. RollCall Error loading index: java.lang.NullPointerException: Attempt to get length of null array. It occured about 20 minutes after being out of range. As of this moment the phone is back to working correctly.

jamorham, 2017-08-03:
Having to reboot the phone implies the bluetooth stack crashed. Not all phones work, some have very buggy bluetooth. There is a list of phones which have had success here: https://github.com/NightscoutFoundation/xDrip/blob/master/Documentation/Phones_For_G5.md

@apeterson22 you should delete the image you posted which contains the qr code barcode as this contains your personal settings data.

apeterson22, 2017-08-03:
Thanks for catching that.. 
I went off that phone list originally when getting the new phone which is why I got the Samsung Galaxy Express Prime (SM-J320A). That was the one that worked for less then 24 hours then wouldn't work.. I might try again with that one, but the Blu seems OK with the exception of the restart.. If I have to do that daily I'll look at the Samsung Galaxy Express Prime (SM-J320A) again. The Blu actually comes with a restart feature (interestingly enough) to schedule phone restarts.. Might look at using that if I run into more restarts being needed and depending on how often.. 
I really appreciate the work and help all of you have given. @StephenBlackWasAlreadyTaken @jamorham @taylorfowler @gwoody1984 @nicking44 ... I will keep you updated on how this phone works. It all else I will connect a Sony SW3 and use that as the collector and just have the phone to upload to Share.

nicking44, 2017-08-03:
On android 7.1.2 Lineage OS for HTC One M9, Xdrip+ is currently working, haven't seen an issue yet.
If any issues pop up (for future note) I'll try the beta on this to see if that resolves it, as well as changing setting around

@jamorham 
Just got xdrip+ setup it's currently operational and is reading every 5 min.

transmitting to pebble works and haven't seen any issues within first 30 minutes of operation.  

@apeterson22 
that's good. Hopefully no more issues pop up. Although if you can I would recommend updating to the latest android version if possible. if that phones is only being used for xdrip it may not be an issue but if it's a daily driver (like mine) I would recommend doing so.

apeterson22, 2017-08-04:
It's strictly for xdrip but I do try to keep it updated as best it can. So
far, just cycling bluetooth on and off if it misses more then 20 minutes
gets it to pick back up but only needed to so that twice while he was in
the water. Thanks again guys and will keep you updated.

nicking44, 2017-08-04:
have you tried using scan constantly, force UI, etc after it started working to see if that helps with those lost readings? Although it could just have lost connection due to interference of some kind, or distance

No problem. happy that we could help out. since Dexcom only supports a few android devices, always glad to help out here when I can, especially since I had my fair share of issues too.

apeterson22, 2017-08-08:
I ended up ditching the Blu phone. 512 Mb of memory is NOT enough for a phone to run on apparently. Xdrip+ was working but would stop every couple hours and restarting Bluetooth or restarting the phone was getting to be a pain. I went back to the J320A just to test once more. I had factory reset it again, since I was getting ready to send it back. This time when i set it up, I made sure I did not restore data. I upgraded OS, installed xdrip+ let it sit for about 24 hours until my son came back, and then started Senor. It picked him up within 10 minutes, but could not write, was only getting Reads. Checked the app permissions and enabled storage permissions (settings>app Manager>Manage permissions>xdrip). Location was already given permissions but the first read from giving storage allow the Battery status and the first Write to occur.  It has been running great since, but last time died about 24 hours in, so I'll keep and eye for a day or two more and report of any further issues.. Thanks..