StephenBlackWasAlreadyTaken/xDrip #146: Alert tones defaulting to silent

bwschulz, 2016-04-09:
I finally got around to installing xDrip and it's working great aside from one issue. The alerts all go off when they are supposed to, but the alert tones do not work. I can 'CHOOSE FILE' for any sound file and the 'Alert Tone' text indicates the sound file I selected, but the alerts have no sound. When I hit 'CHOOSE FILE' again to check the file the dialog box indicates 'Silent'. Interestingly, if I OK the setting of 'Silent' the text still indicates the alert tone I previously selected. The happens for all types of alerts- system, custom, and 'Defautl xdrip'. I can't do anything that makes xDrip make a peep.

This issue exists in v1.0.1 and remains in v2.0.5_2 after I upgraded it.
I'm sending data from a Dexcom G4 with Share, so no Wixel.
Running on LG G3 with Android 5.0.1 and using a Huawei smartwatch.
Please help a newbie to Github and xDrip.

Flohwastaken, 2016-04-13:
I can't help - but can confirm. I'm running Beta v2.0.5_2 and my xDrip doesn't make a peep. Vibration alarms work well. 

I would like to be able to use different sound effects, though.

StephenBlackWasAlreadyTaken, 2016-04-13:
The default alert profile is set to Ascending, this means the first 3 times it "alerts" it will be silent, then gradually increase in volume

If you want it to be audible every time (starting at the first alert) then change the alert profile to medium or loud/high

Flohwastaken, 2016-04-13:
Thanks Steven!

Under "Bg Alert Settings" instead of "Bg Alert Levels" then. Set  mine to high.

bwschulz, 2016-04-14:
Thanks- that did it. Making plenty of noise now.
I think what made me think it was a bug was that I assumed that the "test alert" button would trigger the full alert rather than the first of several and the sound reverting back to silent instead of staying on the one I selected made me even more suspicious.

closed 2016-04-14 by bwschulz