StephenBlackWasAlreadyTaken/xDrip #147: Watchface issues - graph not drawing lines correctly and loses color

Vaughanabe13, 2016-04-15:
I have seen this happen before on different versions of xDrip/Nightwatch but not always. I'm now running the latest beta release of xDrip with the watch features included and now this bug is back in full force.

Steps:
1. I select the watch face on my watch (either xDrip or xDrip Large) and it populates the current data correctly like this: http://i.imgur.com/TM57HCZ.jpg
2. After my watch goes to sleep it randomly switches to the light/white theme for some reason even though I don't have the light theme selected. However the graph still draws correctly, like so: http://i.imgur.com/6ssivTz.jpg
3. At the next 5min interval, the data refreshes from the CGM, but this time all the color is gone from the graph on the watch and there is a straight line drawn from the oldest reading on the screen to the newest reading, like this: http://i.imgur.com/5SzW3AK.jpg
4. From then on, there is nothing I can do to fix the issue, except start over by selecting a different watch face. That line always seems to get drawn between the oldest and newest readings every time the watch refreshes with a new reading. You can see it does the same thing on the xDrip (Large) watch face as well: http://i.imgur.com/2G48Do2.jpg

I have played around with the watch settings in Android Wear, etc. and nothing fixes this issue that I have found. In the past the only way I was able to get around it was by switching to a different build of xDrip/Nightwatch until I found one that worked. The other weird thing is that this does not affect all of my watches the same. Right now it is only happening on my Sony SW3 and not my Moto 360 v1, but in the past it could have been the opposite, or even both watches affected. Right now the Moto 30 works fine and has not displayed the behavior. I don't think it's anything specific with the SW3 is what I'm trying to say, but it does seem to show up sporadically between devices and isn't always consistent. 

Any ideas what this is?

Vaughanabe13, 2016-04-15:
More info: 

It must have something to do with the light/dark toggle in the AW settings. If I toggle the "Dark" button off and back on, it seems to fix the lines issue. Multiple CGM updates go by and the color remains correct. Then if I toggle the "Dark" button off, it does update the screen to the white color scheme, but then the next CGM update will flip the screen black and now there will be no color and the aforementioned line. If I then toggle the "Dark" button back on, it goes back to the correct view and seems to maintain correctly.

starkjoh, 2016-05-04:
I am facing the same issues, besides after some random time I do not recieve any sugar values on the watch. A reboot of the phone / watch does not fix anything. The only way for me seems to be a factory reset of the watch. 
I will switch to nightwatch until this is fixed. Could an update of google play services cause the problem, because its the only thing that was updated recently?