StephenBlackWasAlreadyTaken/xDrip #148: Anyone Built xDrip with one of these yet?

Tatt2edone, 2016-05-13:
I recently stumbled on this after searching for an easier way to solder the hm-10/11 bare boards. They seem to be more abundant that assembled ones. A friend pointed out a BT board at OSHPark, that got me thinking maybe someone designed an xDrip board and they did.

https://oshpark.com/shared_projects/IuYnoGB9

I haven't got the boards yet, they are being manufactured, I am just on the hunt for the parts needed to finish the build when the boards arrive, maybe this set up is over kill or over complicating this, but I am hoping to get the smallest foot print I can and the least amount of parts movement in the case.

Anyway, just looking for some thoughts about future builds with this pcb board

thecubical, 2016-05-21:
I'll be interested to see how it goes, if you can post back thatd be great

Tatt2edone, 2016-06-10:
[image]

I sourced all the parts, my electrical controls friend did all the soldering... Have yet to test it out as I have to still program the Wixel and solder the battery. Looks good though and cost what not all that bad

Tatt2edone, 2016-06-10:
[image]

runeron, 2016-06-20:
Yes, this is exellent, although hard to solder (for me, atleast). Had to buy some solder paste..

He also made a parakeet pcb, which we are gonna test now (he is soldering the first one for me, to see if everything is ok)

https://github.com/mzst123