StephenBlackWasAlreadyTaken/xDrip #149: Is there any iOS app for xDrip?

tmitry, 2016-05-26:
I am looking for an iOS application for xDrip!
Thx

danielschonfeld, 2016-07-05:
I was wondering myself if it would make sense to separate xDrip the visualization and xDrip the data receiver and raw data transformer into a headless xDrip so the setup could be wixel->something with computing power that can run this headless xDrip which would do whatever it needs to the data and let different clients connect to it and then the client can just be a visualization client.  Ie iOS, Android or whatever else maybe even a build it yourself receiver.

I'd be happy to help with PRs in making this happen if @StephenBlackWasAlreadyTaken can help me figure things out about where the crux of the of it all lies and how this works.  (don't have diabetes myself, but am a developer that's really impressed and interested in this project)

JohanDegraeve, 2016-07-05:
Hi, 

I'm trying to setup an ios version for xdrip https://github.com/JohanDegraeve/iosxdripreader
written in Actionscript using FLEX SDK framework and runs on Adobe air.
It means not written in native Java (Android) or objective-c (iOS). I'm doing this because i plan to switch from Android to iOS in September, and because I know this framework. I developed earlier Helpdiabetes ([ios ](https://itunes.apple.com/nl/app/helpdiabetes/id551831317?mt=8)and [android](https://play.google.com/store/apps/details?id=air.be.experit.helpdiabetes&hl=nl)-

Regarding "seperating xdrip visualization/receiver/transformer".
My plan is that the app could be installed on several devices, for the same xdrip device, ie each device tries to connect to the device, but only one can be connected. The connected device (say device A) uploads all data (all = really all = bgreadings, calibrations, ...) to a cloud, other devices then download all data and hence every device has the same view. If device A looses bluetooth connection, then maybe device B takes over the bluetooth connection and takes over the role off uploading, the others to the download.
So every device in the house (android, ios) would have the same data, one of them is uploader, the others are downloader. Or a mix : device A could be connected to the xdrip and upload bgradings, calibrations could be done on device B, ... 
As 'cloud' I use [Google Fusion tables](url), did the same also in HelpDiabetes.

Bluetooth connection seems to work , although less stable than it is with Android I have the impression - but good enough for me.
Currently I'm working on the BgRreading, Calibration and CalibrationRequest classes (all being ported from Android to ActionScript).

If anyone knows the flex framework, I could use help to draw the graphs, never did that before.

whoiswes, 2016-07-05:
@danielschonfeld - I would welcome something along those lines, or merge Nightwatch and xDrip into a receiver/view-only combination app.

Have you seen the Parakeet project @jamorham created?  It's pretty much a xBridge with GSM capabilities.  Might be worth perfecting the xBridge/2/3 concept on the hardware side and leaving xDrip/Nightwatch as viewers.

Personally, I have both xDrip (for myself) and Nightwatch (for my son) running on my Nexus 5 and my battery lasts about 6-8 hours at most (less if I actually USE my phone).  I'd be all over a more efficient way of collecting and uploading CGM data to the cloud and then consuming it via various clients from there.  This sounds like a possible way to accomplish that.

jamorham, 2016-07-05:
@whoiswes There are many different ways this can be achieved I think. Both the Parakeet and Raspberry Pi receivers can consolidate the data to the Parakeet Google App Engine service so both can be read via single URL.

It is also possible to sync xDrip running anywhere with Nightscout via REST and there are many different ways to receive the data from Nightscout. For example I think there is a Nightscout ios viewer. 

On Android you can run xDrip+ as a Nightscout follower using NSClient or if you just want to sync between android handsets then xDrip+ Follower mode allows bi-directional syncing, eg of calibrations etc without needing to run Nightscout.

https://jamorham.github.io/#parakeet

https://jamorham.gitbooks.io/xdrip-plus/content/xdrip-plus-nightscout-viewer.html

danielschonfeld, 2016-07-06:
@whoiswes i agree, it would be nice if we could have a headless part that does all the raw data gathering and run the numbers and create projection numbers and then have 'thin clients' that only present the data.  Which would then become much easier to port to diff platforms ...

Also just as an idea, maybe instead of nightspot's usage of mongo we can also start using influxdb as a backing datastore, which is perfectly suited for this and that db instance can even live right there with the headless instance of xDrip-backend. 

we really need @StephenBlackWasAlreadyTaken input here.

jdgarcia99, 2016-08-22:
Hi All

Any progress on this thread so far?

I would be interested also in having a native iOS app to manage directly xdrip reads

Thanks

JohanDegraeve, 2016-08-22:
I'm progressing yes
- bluetooth connection works, although I still need to test for longer duration to check if it remains stable. That's the part I am most worried bout.
- tested so for with initial calibration, that's ok (comparing with the Android version that is started and calibrated at the same time)
- still need to do the after calibration (ie after 12 hours ...)

- next part will be NightScout upload

As of then I could give you a test version. But it will be very draft

- no alarms yet
- no graphical representation yet of the last 12-24 hours

Maybe 1, 2 or 3 weeks.
I'll update this thread.

jdgarcia99, 2016-08-22:
Ok Johan.

Is it native or are you using adobe framework to programme it?

Let me know if I can help. I have some programming skills as well, or I can help to test

JohanDegraeve, 2016-08-23:
Hi,

I'm using the Adobe framework.
I think it's not used anymore frequently by other developers, and it requires some licenses (Adobe Flash builder, and I'm also using some paid native extensions).
But it will surely help if you can test and give some feedback, and encourage me :)

Hopefully within a week or two, maximum three

Johan

JohanDegraeve, 2016-09-05:
Hi,

i have a version ready but it's still without nightscout upload.
It supports initial calibration, calibration override, calibration, homescreen with current value, calibration request notification, and an always on notification with the value.

I still need to do a final correction which is the linethrough of the measured value, if there's been no more measurements > 30 minutes.

If you want this version then send me your UDID please, preferably via e-mail so i can keep a list of emailaddresses/UDID johan.degraeve@gmail.com

tmitry, 2016-09-05:
> If you want this version then send me your UDID please, preferably via e-mail so i can keep a list of emailaddresses/UDID johan.degraeve@gmail.com <mailto:johan.degraeve@gmail.com>

Hi! Good news!
I am pleased to take part in the test. Thank you for your work and have a great day!

My UDID: 46586bd29698a02ade6df06ab8f8c9ba8e03571c

JohanDegraeve, 2016-09-05:
could you also send me your email address to johan.degraeve@gmail.com ?
so I can keep a list of udid's and owners.

ELUTE, 2016-10-08:
@JohanDegraeve  sent you an email

DennisKampschreur, 2017-07-28:
Am new on GitHub and am interested as well. Has any progress been made on this since oct 8 2016?

JohanDegraeve, 2017-07-28:
welcome,

yes : https://github.com/JohanDegraeve/iosxdripreader

JELCRAWFORD, 2017-08-22:
do I understand that this xdrip version is only functional for G5?  Using a G4 - would love an option for upload on ios.

JohanDegraeve, 2017-08-22:
Also G4 with xdrip it should work, although I'm lacking update with the latest releases of xdrip.
If necessary I can point you to the bridge release that still works for iosxdripreader.

send me your UDID please as described here https://github.com/JohanDegraeve/iosxdripreader

dghinet, 2017-10-22:
Hello! Does anybody tried to connect xdrip with the iPhone 6s?
Do you know if works?

Thank you in advance!

JohanDegraeve, 2017-10-22:
github.com/JohanDegraeve/iosxdripreader