StephenBlackWasAlreadyTaken/xDrip #151: pair wixel-bluetooth with samsung s7 edge

nztianmi, 2016-06-01:
hi i build a bluetooth wixel to use with xdrip, when i try pair with the phone it asks me for a pin, i tried 1234, 0000, 000000 but none of it works. any help? thanks

nztianmi, 2016-06-01:
ok seems it's a problem with marshmallow? doesn't work with my s7 edge and moto x pure. but i have a cheap chinese phone running android 5 i'm so desperate and thought might give it a go, to my surprise it worked!

Cagier, 2016-06-14:
For overcoming issues with your original phone, it may well be related to the issues described in issue #121 so try clicking on that and see if it helps.  If you are happy the problem is fixed then the issue can probably be closed.

closed 2016-06-14 by nztianmi