StephenBlackWasAlreadyTaken/xDrip #152: Xdrip stopped working

RachelMK, 2016-06-09:
Hi All, We run xdrip and it just stopped working at 8:40 Tuesday morning. The yellow light is flashing all the time and it is not finding the transmitter. We've rewritten to the wixel, forgotten and reconnected device, stopped and started sensor in the xdrip app, changed the xdrip app version, unplugged and replugged in the battery and still nothing. Xdrip looks fine but since I didn't build it myself it's hard for me to tell if there's anything loose. It is now stuck on the "needing 2 readings from the transmitter" message. All ideas welcome : )

Cagier, 2016-06-09:
If the yellow light is still flashing all the time then it is not getting a
signal. Are you sure that your G4 transmitter is still transmitting?
Possibly the battery might be gone in that but you should have gotten a
warning first.

If you have changed the app software a few times then you could try loading
xbridge in usb mode and seeing if it is receiving anything. Also be sure
that you are filtering on the right serial number otherwise it will ignore
the readings.

If the transmitter is working and you cannot get anything through then your
wixel could have gone faulty but that would be unusual.

If you know another local xdrip user you could try testing with their
transmitter and vice versa to troubleshoot. (Remember to change the serial
number if you ate doing that.)

If the yellow light stops flashing sometimes then it could be a wiring
problem between the wixel and the bluetooth module.

I'm presuming the xdrip app still shows the status as connected.

Hopefully that gives you a starting point. Let us know how you get on.

RachelMK, 2016-06-09:
Thank you so much for getting back to me!

The yellow light never stops flashing.  Yes the G4 is still transmitting - the dexcom receiver is picking the signal up just fine. It is also a fairly new transmitter that we just set up with xdrip a few weeks ago & it worked completely fine up until Tuesday morning.

Which xdrip app version would you recommend?  I want to make sure I've tried the most appropriate one.  Also - could it have anything to do with the operating system on my phone?  I also wondered about emptying the cache in the app - would that have anything to do with it?

I was a complete "non-techy" before nightscout & am now only very slightly techy : ) so when you say "load xbridge in usb mode" I don't really know what that means ... if I follow the installation video, will this become clearer to me? (this video was invaluable in helping me set up xdrip by the way).  When you say "the right serial number" - you're referring to the transmitter ID correct?  If yes then I haven't changed it since it was set up a few weeks ago so it should be the right one.  There isn't a place to check the transmitter ID within the xdrip app correct?  

Since the light never stops flashing, is there a way to tell if the wixel has gone faulty?  I know you said this would be unusual but unusual is the story of my life : )
 
Prior to restarting the sensor, the xdrip app status did say connected but now it does not.

Thank You!
Rachel
Date: Thu, 9 Jun 2016 09:47:17 -0700

JohanDegraeve, 2016-06-09:
Did you set up the xdrip with the xdrip code (as explained in youtube xDrip (formerly Dexdrip) Wixel Install Instruction Video) or with xbridge code (as can be found here : https://github.com/jstevensog/wixel-sdk/blob/master/apps/xBridge2/xBridge2.wxl)

In the second case (xbridge2) you can indeed configure the transmitter id by using the android app - then you need to set Hardware Data source to xbridge Wixel.

Cagier, 2016-06-09:
If the yellow light is flashing rather than steady then you are probably using the original xdrip/Dexdrip code on the wixel.  However, if you load the xBridge2 code referred to above, you can then connect a USB cable to the Wixel and use another program to read the output on your computer which is helpful for troubleshooting.

I wouldn't worry too much about which app version you are using.  They should all get the signal but latest one is probably better.  When you refer to xDrip app it is a little confusing as the physical xbridge is often referred to as an xDrip but the program that runs on that (as described by John above) is obviously very different to the xDrip app that runs on your phone.

The fact that you said that you have "rewritten to the wixel" implies that you have loaded the xdrip (dexdrip.c) code onto the wixel.  That contains the Transmitter ID (which I was referring to as serial number) and you seem to already understand that.  I was just concerned that if the person who built your xdrip loaded that code on for you then you might have overwritten it if you didn't put in the correct Transmitter ID in dexdrip.c when you reloaded the code.  Again, as John says, if you use the xbridge2 program instead, you can simply program the Transmitter ID from the phone app instead.

The only other concern is that you say it no longer says connected.  If you are not connected then you will not get any readings.  There have been some ongoing reports of bluetooth connection problems and missed readings.  Coincidentally my own (son's) xdrip stopped working a few days ago with my own phone (a motorola) but still works with another phone.  So there may have been some sort of android update or something that has stopped the phone from working properly. 

On that basis, maybe another option is to try loading the xDrip app onto another phone and see if you can connect with that?  Be careful though as the two phones may then "fight" for the bluetooth connection so you will need to "forget" it afterwards.  Safest bet is to switch off bluetooth on your existing uploader phone while you are testing another phone.

So there are a couple more suggestions for you.  The USB debugging is a little complicated if you are non-techy so maybe try the other phone first?  Then just try the xbridge app and set the Transmitter ID accordingly.  Finally, if all that doesn't work then look at using PuTTY or some other console program to view the Wixel output as described in the xbridge2 documentation.

I hope you get it working soon!

RachelMK, 2016-06-09:
I do yard xdrip - formerly dexdrip

Cagier, 2016-06-09:
Well if you have "reloaded" that code then be sure that the code you are loading includes the right transmitter ID.  As per previous post, try with another phone first and then move on to troubleshooting with xbridge2 app.  At least those are my suggestions.  Others may have a different opinion or experiences...

RachelMK, 2016-06-10:
Thank you. I don't have another android phone to test it on so I will likely just try xbridge. Also the system status now says connected but I still have the same messages I.e. "Bg reading unavailable - xdrip data collections service is running" on the home screen of the phone and "please wait, need 2 readings from transmitter first" in the xdrip app. Perhaps a dumb question but does xbridge mean that the G4 transmitter (without share - we don't have that in Canada) can be used with iPhones?

RachelMK, 2016-06-10:
Can you tell me exactly how to load the xbridge2 code? Is there a video as with the xdrip?

JohanDegraeve, 2016-06-10:
@RachelMK it's a lot easier with xbridge2. You download https://github.com/jstevensog/wixel-sdk/blob/master/apps/xBridge2/xBridge2.wxl
Connect the xdrip to pc, start Wixel configuration utility.
Top right corner, click "open", and load the xbridge2.wxl
Write that to the wixel.
Now disconnect.

Open the android app, set hardward data source to xbridge, you should now have the option to set the transmitterid in the android app

RachelMK, 2016-06-10:
Thank you - & this will work with my xdrip rig?  I thought I read somewhere that the wiring on the rig had to be a bit different for this to work.  I will try it tonight.
Rachel

Date: Fri, 10 Jun 2016 00:15:02 -0700

Cagier, 2016-06-10:
The wiring is different for the battery voltage measurement functionality and also some power saving features with the Bluetooth module but you should still be able to get the USB output that you need with the traditional xDrip rig that you have.

RachelMK, 2016-06-10:
I don't see anywhere on that page that I can click "download"
Date: Fri, 10 Jun 2016 00:15:02 -0700

JohanDegraeve, 2016-06-11:
If you click on "raw", then the browser opens the page, then you can save the page on your pc.

RachelMK, 2016-06-12:
OK wrote xbrdige to the wixel, put the transmitter id in the xdrip app, restarted sensor & now the yellow light isn't blinking at all
 
Date: Fri, 10 Jun 2016 00:15:02 -0700

JohanDegraeve, 2016-06-12:
Hi,

I also just installed this latest version and it seems indeed the yellow led is not blinking anymore at all.
I assume the developers change their mind from time to time regarding this blinking :) probably a way to save energy.

Does it work now ? do you receive updates on the app within 5 minutes ? For me it works, i receive updates, but the led is not blinking. It might ligth up for a few seconds just before receiving the transmitter signal - means the wixel has woken up and is now listening. But not sure I didn't look at it :)

Cagier, 2016-06-12:
Yellow Light:
Yes, the Yellow light behaves differently with the xbridge software rather than the xdrip software.  I'm not sure how long ago since I downloaded my version of the xBridge software but the yellow LED stays on for about a minute with the reading happening somewhere in the middle.  I think this because there is a bit more of a conversation between the xDrip and the phone so it comes on, waits for the signal and then goes off again after a little while.  (I think there was also an option to switch off the lights  from flashing in the code so current version my default this off.)

Testing:
Anyway, if you are testing then tell it that you started the sensor over 2 hours ago and it should start looking for readings straight away.  As it needs two reading before looking for a calibration it will take at least 10 minutes for the first message about waiting for readings to go away (and then look for a double calibration).  If it is still waiting for readings after 10 minutes then you probably still have a problem.

Logging:
Hopefully it is working now but if you still have a problem then you should try connecting your xbridge device up to the USB port and seeing what information you can get from the wixel log.  You need to connect the wixel to a computer using USB, and it will appear as a serial (COM) port. (take note of which port it is by looking at the device manager in your control panel or maybe you can use the wixel configuration utility to figure out which port it attaches as. Then run PuTTy, Hyperterm or any other serial terminal program to connect to the COM port the wixel is using, You should also have an option in the program to store a session log into a file. You can press the S key to get a status at any time. This will show you the version of code, and a few other things.  Sorry if that all sounds a bit complex but see how you get on.

Potential Issues:
Can you also confirm that the xDrip app status says that the device is connected?  I think you are saying that it is and so that would rule out any bluetooth problems.  Then either your problem is wiring between the wixel and the bluetooth module or the wixel.  However, as the previous version of the xdrip that you were using does not really know if it is connected to a bluetooth module or not, the assumption is that it was not a wiring problem (because the yellow light should still have stopped flashing when it gets a signal).

Diagnosis:
Ultimately if it is still not working then the logs from logging exercise above will hopefully shed some light on what the problem is.  However, if everything appears like it is working but the device is just not receiving the G4 signal (and your other receiver is) then it is probably a faulty wixel.  But hopefully not...

So let us know how it all works out!

RachelMK, 2016-06-13:
Thank you all so much for you help!  It really is amazing that you all volunteer your time to support such an important technology for people living with T1D.  I'm still having issues so another amazing volunteer who actually built my xdrip is allowing me to mail it to him so he can take a look.  I will post here & let you know what the issue was.  Thank you again - I've learned a lot about how to trouble shoot the xdrip.
Rachel

Date: Sun, 12 Jun 2016 15:24:38 -0700