StephenBlackWasAlreadyTaken/xDrip #153: BT connection worked before, now won't pair

Vaughanabe13, 2016-06-23:
I unpaired and did a "forget" on both the xDrip app and my G4 Share receiver. Now when I try to do a new BT scan from within xdrip, nothing comes up in the scan results. If I go to my phone's bluetooth menu and do a search, DEXCOMRX shows up there, so I know my share receiver is visible. I can also see the bluetooth logo flashing on the G4. Restarted my phone, tried toggling bluetooth on and off, tried toggling airplane mode, tried uninstalling/reinstalling the app. I'm on beta version 2.0.5_2. Also I double and triple checked that my serial number was entered correctly and the communication type is set to Dexcom Share. I appear to be doing everything correctly but the app simply will not connect anymore. Any suggestions on what to try?

Vaughanabe13, 2016-06-23:
I just uninstalled and went back and installed an older version of xdrip that I had on my phone, version 1.0.1 from 7/26/2015 and then I ran through the normal connection steps and this time the bluetooth scan came up with the DEXCOMRX successfully.

Then without changing anything else I tried uninstalling 1.0.1 and reinstalling 2.0.5_2, re-entering the serial number and the bluetooth scan still won't return any results.

Vaughanabe13, 2016-06-28:
Any ideas? I can't get any version of xDrip with Android Wear integration to read my Share receiver anymore. Even though it was working previously just fine. When I press "SCAN" in the BT Scan menu, it just sits there and never comes up with any results.

StephenBlackWasAlreadyTaken, 2016-06-28:
Go into settings, Apps, xDrip, permissions and see if location is enabled

Vaughanabe13, 2016-06-28:
That's it! Permission was not enabled. It's very odd, when I install the app I get no option to accept permissions, the app just says there are no particular permissions necessary. I'm guessing this is a change with Marshmallow but I haven't had to manually adjust permissions for other apps yet. Either way, once I enabled location the BTScan worked perfectly and now I'm up and running again. Many thanks!

closed 2016-06-28 by Vaughanabe13