StephenBlackWasAlreadyTaken/xDrip #159: [G5] Beta -Transmitter Batter: Not available - Need 2 readings, no option to do so

nicking44, 2016-07-30:
So I posted this over on the Facebook group a couple hours back but still no solution to it, posting here just in case someone can help me out. I'm getting "please wait, need 2 reading". Entered in the same info that worked before and my Dexcom receiver got data so the info is correct. images provided are some errors that I'm encountering. Any help would be appreciated. 
[image]
[image]

nicking44, 2016-07-30:
as well this is currently my only Bluetooth device attached and Location is enabled in the permissions.

jfbureau, 2016-10-10:
Did you get this working?  I have the same phone with the same issue.  I am transferring from a MotoX, where it worked fine.

nicking44, 2016-10-10:
@jfbureau No I didn't I'm going to wait to try it again, when I get a new transmitter. May I ask did you receiver die when this started happening, because that's what I believe what caused this issue or not. I'll update if my new transmitter fixes the issue or if I got it fixed before changing.

borijess, 2016-10-31:
I would love to figure this out. I have a Oneplus 3 android phone that has the same issue. I am coming from the Note 7 that worked great. Would love to know what can fix this.

nicking44, 2016-10-31:
@borijess I'm going to ask you as well did you dexcom receiver "die", are you able to connect the 2 together?

borijess, 2016-10-31:
The receiver works fine but I don't use it. I went back to my note 7 and xdrip working fine again. Does not work with oneplus 3. Any idea why cause in have to return note 7.

nicking44, 2017-08-04:
@borijess  I know this is an old post, have you gotten it to work on the other device yet?

apeterson22, 2017-08-08:
@borijess  Did you get this working on your device? I had some similar issues with a device, and found that storage permissions wasn't letting the app write. Once I gave xdrip+ the permissions to storage (also gave it permissions to everything I could this phone is strickly for xdrip) took about 3-4 reads and it was able to start writing after a collector restart. Just a thought on something to possibly check.