StephenBlackWasAlreadyTaken/xDrip #160: Instructions for El Capitan

square-eyes, 2016-08-22:
I tried setting up the app from the instructions [https://github.com/StephenBlackWasAlreadyTaken/wixel-xDrip#on-a-mac](here) for mac and ran in to a couple of issues.

Running the curl command for sdcc fails with a whole lot of dir errors - even as root e.g. cp: /usr/bin/as2gbmap: Operation not permitted.

I installed it using brew and that worked OK.

Then when I try to compile the app I get the following errors...

Compiling apps/dexdrip/dexdrip.rel
apps/dexdrip/dexdrip.c:85: warning 158: overflow in implicit constant conversion
apps/dexdrip/dexdrip.c:85: warning 158: overflow in implicit constant conversion
apps/dexdrip/dexdrip.c:85: warning 158: overflow in implicit constant conversion
apps/dexdrip/dexdrip.c:85: warning 158: overflow in implicit constant conversion
apps/dexdrip/dexdrip.c:86: warning 158: overflow in implicit constant conversion
apps/dexdrip/dexdrip.c:86: warning 158: overflow in implicit constant conversion
apps/dexdrip/dexdrip.c:86: warning 158: overflow in implicit constant conversion
apps/dexdrip/dexdrip.c:86: warning 158: overflow in implicit constant conversion
apps/dexdrip/dexdrip.c:213: warning 158: overflow in implicit constant conversion
apps/dexdrip/dexdrip.c:213: warning 158: overflow in implicit constant conversion
apps/dexdrip/dexdrip.c:213: warning 158: overflow in implicit constant conversion
apps/dexdrip/dexdrip.c:213: warning 158: overflow in implicit constant conversion

Lastly, and also probably unrelated... am I right in assuming the xDrip needs to be made before the app is put on the Wixel? Or can I do the app stuff first?

Cheers!

JohanDegraeve, 2016-08-22:
giving the same answer as in https://github.com/StephenBlackWasAlreadyTaken/wixel-xDrip/issues/51

You can install the xbridge version : https://github.com/jstevensog/wixel-sdk/blob/master/apps/xBridge2/xBridge2.wxl
No need to compile yourself, this was needed because the transmitter-id was hardcoded in the xdrip code. This xbridge code it's not necessary.
You can install it his directly via the Wixel configuration utility.

Doesn't matter if you've build an xdrip or xbridge. The sofware will run.
Advantage is that you can set the sensor id via the Android app, so you don't need to do the make stuff.

Once you have the Android app installed and bluetooth connection, then go to settings, Verify dat hardware data source is set to "xbridge Wixel", if that's not the case just change it.
Then you will see the option to set the sensor id (or Transmitter id)

square-eyes, 2016-08-22:
OK thanks I did read that, but couldn't see what Android app it was referring to since the transmitter ID input doesn't appear until xBridge Wixel is selected, as you say (I had data source to Bluetooth Wixel).

Also, what about setting up the Wixel before the xDrip is built? Any ideas why the Polou doesn't show up in the utility on OSX (I haven't built anything yet just wondering if I can flash this before I do). Maybe it's an El Capitan thing and I have to try Windows.

JohanDegraeve, 2016-08-23:
I'm not sure but I think yes you can program the wixel before the xdrip is built.

I have no experience with OSX. Maybe better to install Virtualbox with a Windows guest on your Mac

closed 2016-10-17 by square-eyes