StephenBlackWasAlreadyTaken/xDrip #161: About to use for the first time

square-eyes, 2016-08-23:
I have made my xDrip and it's picked up by the Android app, which tells me to now activate my transmitter, but a couple of things are holding me back. I'd love some feedback before I waste my first ever sensor and a month of transmitter.

1. Is this proof it works? - Or only that the bluetooth module is paired. Can I otherwise test it without inserting a sensor and activating my transmitter?
2. Secondly, I ordered a [BLE HM-10 from Aliexpress](https://www.aliexpress.com/item/AT-09-BLE-Bluetooth-4-0-Uart-Transceiver-Module-CC2541-Central-Switching-compatible-HM-10/32472162107.html?spm=2114.13010608.0.74.0YxHAN) but it looks like a non branded one. While it shows in the app and allows me to select it, am I at risk of a fail point here once I take the plunge (so to speak)
3. Thirdly, my battery hasn't arrived. So I connected a 500ma USB charging cell to my charging unit, and tested 4.2 volts across the battery terminals (where you would connect the battery to charge it. I got just over 5 volts on where you'd connect the Wixel, so temporarily soldered the Wixel to the batt terminals. Am I crazy in doing this? Connecting the usb cell powers up both the BLE and the Wixel. I actually quite like being able to use the external cell like this.
4. Lastly I may want to re-solder and tweak the box. Can I do that after I trial it with the sensor etc, without ruining anything (I assume so but being sure)?

Obviously, if I need to order more parts it could take weeks, and I'm doing this whole thing on the cheap so would be mortified if I lost a sensor or wasted my transmitter, to the point I may as well cough up for a receiver if it turns south. That I can't afford.

JohanDegraeve, 2016-08-23:
Are you using Dexom G4 ?

square-eyes, 2016-08-23:
Transmitter yes, no receiver.

JohanDegraeve, 2016-08-24:
A Dexcom G4 is 'activated', as soon as you take it out of the box (there's a magnet in the box). Then it will run and start consuming battery. It lasts for more than a year normally.

"Starting a sensor" only means the algorithm on the android app starts (in case you would use a Dexcom receiver it's the algorithm on the receiver). But there's no feedback to the transmitter.

It is important that when you start the sensor, you set the correct timing. For example say you insert your sensor today at 8:29 but you start the sensor on the app only at 12:00, then you should set the start time on the app correctly on 8:29. In that case, just 10 minutes later you would be able to calibrate (it needs two readings and there's a reading every 5 minutes).
If you would start the sensor on the app say at 9:00, then you would still need to wait till 10:29 + 10 minutes before you can calibrate the app. 

You can resolder the the things, don't "stop" the sensor on the app, just keep it running. It won't receive anything while you're tweaking but the influence should be low i think.
You could also stop and start again the sensor, just make sure you set the correct timing.

I think normally you can use the wixel/bluetooth module without charger and battery of you connect the USB to the wixel.

Sorry I don't have all the answers.

square-eyes, 2016-08-24:
Ok thanks, all good advice :) I inserted my first sensor and activated my sensor. Now I wait!

square-eyes, 2016-08-25:
OK well, I'm up and running. And lost for words. It's been a long road getting here. I'm beside myself and can't stop looking at it.

What a great project. Thanks for the time and all the help to everyone involved.

square-eyes, 2016-08-25:
I should make some notes in case it helps anyone.

1. Since my battery hasn't arrived. I have successfully connected my portable USB cell charger to the charging port of the charger. I used that because it has micro USB whereas the Wixel (if I'd have connected it direct to there) has mini USB. Also when I get the battery I can just add it. It seems the Wixel works just fine connected direct to USB power, or the batt terminals of the charger, or the device terminals of the charger. I moved the wires to the correct location as per the instructions, and have no issues, even though it measured 5v.
2. My Chinese BLE works. I was mortified when I found I didn't have the branded one. And found people on Arduino forums saying the clone wasn't working for their project. There's even a guy here somewhere that had problems with his. The one I have linked to above seems to work!
3. I had a stressful hour, very late at night after the initial 2hrs warm up where it wasn't working. I thought I was done-for (at least my wallet a la wasted sensor). But it was the Wixel software which I had compiled myself. I re-loaded the pre-compiled one linked on here and it worked. A late one for me at 3am, but I went to bed the happiest man alive and woke up to glorious data. I'm already learning so much I could cry.

JohanDegraeve, 2016-08-25:
Indeed it is an amazing project.
You're just starting to use it now, it will only get better. Note that the first 1-2 days of sensor usage there may be a bit more deviation on the results.

square-eyes, 2016-08-25:
Is that for every sensor?

JohanDegraeve, 2016-08-25:
Yes, actually the algorithm (the android app but also the Dexcom receiver) is taking this into account. That's why it's important to set the correct timing when you start the sensor - or better to start the sensor at the moment you insert it.

But you will probably still see some deviation which is higher the 1st .. 2nd day
There are some experts here that even adapt the algorithm to their personal experience.

When using Dexcom and restarting if after 7 days but keeping the same sensor, this may even lead to more inaccurate results because the receiver thinks it's a new sensor while it isn't.
See : http://www.tudiabetes.org/forum/t/restarting-the-dexcom-sensor-what-you-should-know/14611

square-eyes, 2016-08-25:
Great thanks for the info. Yes I'm cross referencing with my glucometer quite regularly until I get a feel for the G4 data.

square-eyes, 2016-08-31:
Regarding restarting the same sensor. Is it possible to add a setting in the app to mark it as not being new so that the day 1 & 2 compensation doesn't occur? I'm pretty new but will eventually get in to the code. Although I'm not on that level yet. Looking forward to contributing though. I'm happy to test new releases. Actually what is the feature request process?

square-eyes, 2016-08-31:
Actually, on xDrip, do you even need to stop the sensor after a week?

JohanDegraeve, 2016-08-31:
no you don't need to stop the sensor.
If you want to develop : fork the code, make and test the changes, push them, and make a PR (pull request).
It requires some git(hub) knowledge

To make a feature request : i think you can create an issue.

square-eyes, 2016-08-31:
Cool thanks.