StephenBlackWasAlreadyTaken/xDrip #162: Wixel Surgery - Broken 3v3

square-eyes, 2016-08-28:
Disappointed to report the 3v output eyelet ring has pulled out of my Wixel board.

I'm not using pins and had to re-seat the solder joint as the wire was not sitting correctly. Removing the solder resulted in the eyelet pulling clean off the board with no way to reconnect the 3v wire.

Can any of the other terminals be programmed or can I otherwise replicate the function of that terminal, or can the BLE draw power from the VIN?

What would be the consequence of the latter?

garidan, 2016-10-07:
The BLE can draw power from the 3V in, or you can use the XBridge schema and draw power from and I/O pin

square-eyes, 2016-10-10:
How would I use the XBridge schema? - By that do you mean edit the source to controll another I/O in place of the 3v3?

garidan, 2016-10-11:
I mean just that, but you don't need to change source, it's already done in XBridge. look here, there is a PDF with documentation and schema. It just uses a I/O pin to drive current to wixel. Than you can configure it to keep it always on or to switch it off when not needed.
https://github.com/jstevensog/wixel-sdk/tree/master/apps/xBridge2

garidan, 2016-10-11:
You just use that wxl to reprogram wixel. PDF is well done, read it then if needed ask me here

closed 2016-10-17 by square-eyes