StephenBlackWasAlreadyTaken/xDrip #169: I can't get this thing working HELP!

theKarmaking, 2016-11-12:
Hi, my daughter is 5 and we are trying to get this going asap. i've read the instructions, downloaded software and can't figure this out. The Wixel config utility recognizes my device. I click install app and go to the xdrip software and there is no wxl file. what am i doing wrong.

JohanDegraeve, 2016-11-12:
Hi,
use the xbridge software, just download from here to your pc and install with the Wixel configuration utility.
https://raw.githubusercontent.com/jstevensog/wixel-sdk/master/apps/xBridge2/xBridge2.wxl

When you start the Android app, configure your device as "xbridge" (even if it's an xdrip), this allows you to configure the tranmsitter id

Johan

theKarmaking, 2016-11-12:
Posted too soon, got the app installed using a video, (Props to Andrew!) now trying to sort getting it connected. Thanks

detroitdr, 2016-11-24:
what video?  can you provide a link please. Thank-you

theKarmaking, 2016-11-29:
https://www.youtube.com/watch?v=YVGqzYRQXkk is the link it helped me out!

detroitdr, 2016-11-30:
Thank-you. I'll check this out tonight.Greatly appreciate it!