StephenBlackWasAlreadyTaken/xDrip #177: proper watch face for pebble time round?

Mendanbar, 2016-12-25:
I just tried to load the xDrip watchface to my pebble time round, and it looks to be incompatible.  I'm guessing the xdrip app is sending the regular pebble time version only?  I'm using the latest beta (as of 12/24/2016).  For now I can use one of the other cgm watchface options, but I'd rather use the xdrip face since it will still work without internet (our internet experiences outages frequently).  Is there a version for the pebble time round in the works?

Thanks.

sochs94, 2017-03-01:
You can use xdrip+ with the pebble time round workaround from the last beta

pljcgm, 2017-12-30:
Can someone tell me how to setup the pebble time round workaround?

sochs94, 2017-12-31:
First download xdrip+ with this link: 

https://github.com/jamorham/xDrip-plus/blob/master/Documentation/images/download-xdrip-plus-qr-code.png

Install that
Then download the Pebble Round with this one:

http://bit.ly/pbw-xdrip-round

That file, once downloaded, should open in the Pebble app for installation, go ahead and do that. 

Then go into xdrip+ settings/smartwatch features/pebble integration, and turn on Pebble Watch integration. On your watch select the xdrip watchface, and after a few seconds your bg should load. Hope I didn't miss anything, it's really not that difficult. Good luck!

pljcgm, 2018-01-03:
Cool, thanks :). Can I install that version of xdrip without loosing all my data from my current version of xdrip?

jamorham, 2018-01-03:
@pljcgm You would need to export your data first using the xDrip export menu and save settings feature to save your settings to sdcard so they can be reimported when you install xDrip+

@sochs94 What is the difference with the round watchface you linked to vs the round one which is bundled with xDrip+ ?

sochs94, 2018-01-03:
I wasn't aware that there was one bundled with xdrip+. Thanks man!

sochs94, 2018-01-03:
Actually, just looked and I don't see a round watchface option there... am I missing something?

jamorham, 2018-01-03:
Just install the Pebble Classic Trend or Color Trend watch face. They are universal and work on every different pebble model. It isn't quite as pretty on Round as it is on for example Pebble 2 as the screen real-estate is more limited

pljcgm, 2018-01-03:
@jamorham That's actually not true. At least not for my version of xdrip. In the pebble app it says that the watchfaces are not compatible with my pebble.

@sochs94 I got it to work with the workaround. Thank you so much! :)

jamorham, 2018-01-03:
@pljcgm which version of xDrip+ are you using? I have tested the mentioned watchfaces with pebble round and they are loading fine here

pljcgm, 2018-01-04:
@jamorham I'm not sure unfortunately. I'm really new to this ;). Just downloaded it from https://github.com/NightscoutFoundation/xDrip/releases .