StephenBlackWasAlreadyTaken/xDrip #179: broadcast locally

danpowell88, 2017-01-30:
How exactly does this work, is there anyway to have another app on the phone hit a local http endpoint and receive the latest values?

Looking to do this as a hack around Garmin Api only allowing communication via http and wanting to have it work when I can't directly connect to nightscout.

naftis, 2019-01-30:
@StephenBlackWasAlreadyTaken, this is the first search result when googling "xdrip broadcast locally". How can developers integrate into xDrip using this? Some information could be here; is there a example source code from an app I can use? Is there documentation somewhere? Would definitely appreciate some help!

Rytiggy, 2019-01-30:
Hey @naftis you can access data on xDrip by turning on the xDrip Web Service to do that follow the steps below: 
 Navigate to `Settings` -> `Inter-App settings` -> `xDrip Web Service` -> `ON` 

then once you have that enabled you can view the blood sugars by going to this url http://127.0.0.1:17580/sgv.json on your phone.