StephenBlackWasAlreadyTaken/xDrip #180: Feature Request - Auto Snooze when treatment is input

Andyp2010, 2017-02-04:
Two features would be fantastic - if it would snooze automatically when inputting treatment rather than having to do both, especially helpful on a night.

Also if you can snooze it normally from the android watch - we haven't found a way yet.

Fantastic app, thankyou for advancing the field so much!

catslover32, 2017-02-04:
I agree, this feature would definitely be appreciated

amlynek, 2017-06-26:
I would like to see the snooze button somewhere easily accessible - maybe under the 'speak readings' option (top right menu in xDrip), or something semi transparent overlaying the entire xDrip app when triggered. Right now it is buried in the main menu and getting to it at night when you are half awake is quite the challenge. Just a suggestion. Thanks for reading.