StephenBlackWasAlreadyTaken/xDrip #182: A Question

l1106397, 2017-03-05:
Hello,

Thanks for posting this!
I was wondering how you get this application ID?
"applicationId" : "d89443d2-327c-4a6f-89e5-496bbb0317db"

Is it provided to you by Dexcom?

osa1, 2022-06-23:
Ping @StephenBlackWasAlreadyTaken. This application id is used by a few different apps today, probably because no one knows how to get a new id. Could you help us out with this please?

Emma-Black, 2022-06-23:
This account id was grabbed off a man in the middle attack where I had an iPhone connect through my computer to the internet (using charles proxy) with the wrong certs accepted, this allowed me to see the communication back and forth between the dexcom server, and the iphone that was running the dexcom app

I assume it was hard coded into the app, so I just copied it straight out of there.

heres the details on the share1 endpoints:
https://gist.github.com/StephenBlackWasAlreadyTaken/adb0525344bedade1e25

Now, if you are having trouble with share1, its because I think dexcom has moved everyone over to share2 (or something)

I have not looked at the endpoints or MITMing it myself but someone helpfully wrote up a comment with all the details they gathered on the share2 endpoint here:

https://gist.github.com/StephenBlackWasAlreadyTaken/adb0525344bedade1e25?permalink_comment_id=4001119#gistcomment-4001119

but the account Id is still the same interestingly enough!

osa1, 2022-06-23:
Thanks for the details @Emma-Black, and thanks for documenting the API. I can't express how helpful this has been.

Emma-Black, 2022-06-27:
Im really glad it helps! sorry Im not around too much these days!